OrcaSlicer/OrcaSlicer #149: Feature request: Time when finished

leofaoro, 2023-01-09:
Hi, woudl it be possible to add a field in the device status showing the time when a print is expected to complete? thanks

SoftFever, 2023-01-10:
Can you provide more details?
I believe both Bambu and Klipper already support this feature

leofaoro, 2023-01-10:
Sure. Just show the time when the print is expected to be completed, in addition to the time left. This is in the "printing progress" box in the "device" tab. See below for mockup

[image]

ymilord, 2023-03-22:
In Superslicer it had an option via M117 to display the time remaining. Is that still possible?

closed 2023-11-25 by github-actions[bot]