OrcaSlicer/OrcaSlicer #153: Variable mesh slicer parses real world positions

Prutsium, 2023-01-10:
BambuSlicer (SoftFever)1.4.2

When using a variable mesh size based on the print size on the plate the slicer parses wrong values.
It seems it uses real world values based as in the example a 300x300 printer is been used (RatRig vcore3 300mm) and i have 4 plats prepared so its a 2x2 layout see screen shot.
[image]

So plate 1 gets 0-300 on X+Y parsed all is ok but plate 2 gets 300-600 X and 0-300 on Y parsed.

In slicer start Gcode i have the following to get the mesh size parsed (this all works very well with Super/Prusa Slicer)
MESH_CONFIG X0={first_layer_print_min[0]} Y0={first_layer_print_min[1]} X1={first_layer_print_max[0]} Y1={first_layer_print_max[1]}

And here are the lines from the gcode files of these 4 plates:
1: MESH_CONFIG X0=7.53444 Y0=15.5143 X1=263.444 Y1=296.579
2: MESH_CONFIG X0=364.939 Y0=5.53464 X1=657.669 Y1=288.3
3: MESH_CONFIG X0=66.3816 Y0=-326.496 X1=240.681 Y1=-132.059
4: MESH_CONFIG X0=377.405 Y0=-309.519 X1=645.067 Y1=-90.3092

3mf attached (zipped due Git)
[ENC2.0_3.1.zip](https://github.com/SoftFever/BambuStudio-SoftFever/files/10379267/ENC2.0_3.1.zip)

SoftFever, 2023-01-10:
Yes, it's a known bug.
At the moment, only the first plate got the right size.
See issue #37

Planned to fix it in the next release.

SoftFever, 2023-01-22:
fixed

closed 2023-11-25 by github-actions[bot]