OrcaSlicer/OrcaSlicer #154: Part cooling logic seems to be broken

H4kunaM4tata, 2023-01-10:
**Describe the bug**
I have this option disabled because I don't wanna part cooling on at all for the first 2 layers.
At least that is what I expect it to do.

[image]

But that fully disables the part cooling for all the bottom and top layers which is wrong.

**To Reproduce**
Steps to reproduce the behavior:
1. Disable keep fan always on
2. Slice
3. Check the fan speed color scheme

[image]

If I do enable keep fan always on, then it does work as expected.
Or I am missing something or something doesn't seem right.

[image]

Thank you

SoftFever, 2023-01-10:
I haven't dug into this part yet. 
My hunch is that when you have `keep fan always on` off, the slicer will turn off the fan speed if the auto-calculated fan speed is below the min fan speed. In your case, the min fan speed is 100%.
So when the calculated fan speed(based on layer time) is below 100%, the fan will be turned off.
Sounds like an expected behavior to me.

H4kunaM4tata, 2023-01-10:
@SoftFever I confess I am lost here with that logic.
I will use what works as expected for me, since this is an expected behavior, I am closing this.
Thank you

closed 2023-01-10 by H4kunaM4tata