OrcaSlicer/OrcaSlicer #156: Add Extrusion System Benchmark similar to CNCkitchen's to calibrate max volumetric flow

linyongzheng, 2023-01-10:
**Is your feature request related to a problem? Please describe.**
Before performing LA/PA calibrations, would be great to have tool to improve finding max volumetric flow built into slicer.

**Describe the solution you'd like**
In the calibration menu, creates a test pattern with the user-inputted temperature sweep parameters. CNCkitchen created an excel tool for this (which has also been ported to a webpage) but would be great to have this built into slicer to use the majority of the custom start codes.
(https://github.com/CNCKitchen/ExtrusionSystemBenchmark)

**Describe alternatives you've considered**
Can use the excel sheet or the web page port (https://hotend-flow-tester.netlify.app/) but I don't do it very often because I try keep track of different machine's start & end gcodes to prevent potential crashes.

psiberfunk, 2023-01-11:
I agree this is something that would be useful.  What's nice about CNC kitchen's approach is that it really helps you find the linear and non-linear regions for the given hotend/extruder/temp combination you've chosen.

linyongzheng, 2023-01-20:
Another possible implementation: https://www.printables.com/model/318350-bambu-slicer-max-flow-test-tower-for-x1c

SoftFever, 2023-02-26:
Thanks for the suggestion.
It is implemented in 1.4.5.
Closing it now.

closed 2023-02-26 by SoftFever