OrcaSlicer/OrcaSlicer #157: Feature request: Jerk / SQV settings for infill

Prutsium, 2023-01-10:
Can we get also a Jerk / SQV settings for Infill here?
[image]

As i run Klipper on a fast printer and print alot of Gyroid i want to increase the SQV for infill. In superslicer i used the variable extrusion role but as this isnt possible here it would make sense to bring it under the Jerk settings.

Thanks

cron410, 2023-01-13:
I fond this macro set somewhere online years ago. Might help you. 

[fast_infill.cfg.txt](https://github.com/SoftFever/BambuStudio-SoftFever/files/10407655/fast_infill.cfg.txt)

Prutsium, 2023-01-13:
> I fond this macro set somewhere online years ago. Might help you.
> 
> [fast_infill.cfg.txt](https://github.com/SoftFever/BambuStudio-SoftFever/files/10407655/fast_infill.cfg.txt)

I know that route. Was using that route long ago but since Slicer has all the info and Prusa/Superslicer can do this themselves based on the parameter extrusion role just 2 lines would be needed.
[image]

But it would be great to have it in the uniform way of BambuSlicer (Softfever) as the Jerk / SQV settings in my 1st post.

SoftFever, 2023-01-22:
Good idea.
Added in b7bdb65f

Prutsium, 2023-01-22:
Thanks! Makes the slicer more usable for Klipper users!

The-Futur1st, 2023-01-23:
@SoftFever Please consider to add same parameter for acceleration just to be consistent. I have also thought about acceleration control for bridging perimeters. I did not make tests of that, but from my predictions it can be helpful with filled filaments like PP-GF30 or similar(such filaments usually very difficult with bridges and force you to print bridges with very low speeds 15-20mm/s)

closed 2023-05-06 by SoftFever