OrcaSlicer/OrcaSlicer #161: Enhance Ease of Removal of Pressure/Linear Advance (Line Method) print

rlabusiness, 2023-01-11:
**Problem**
I love the simplicity and speed of the Pressure/Linear Advance calibration print. However, removing it (using a scraper) results in many statically-charged lines and numerical digits literally bouncing/springing from the build plate many feet away in multiple directions. They also end up sticking to my scraper, the back of my magnetic PEI build plate, etc.

**Desired Solution**
Ideally, I'd love to scrape this print off the bed without removing my build plate and holding it inside my trash bin before scraping it off. Following lead of Bambu's calibration print pattern, could this calibration include a few additional lines to connect all of these small pieces together?

**Example**
As a potential approach (illustration below if helpful, with red lines as an example):
1. Draw two vertical lines connecting all of the left or right ends of all horizontal calibration lines
2. Underline all digits in each row, and connect these underlines to the right-most vertical line from #1

[image]

SoftFever, 2023-01-12:
The digits are the most painful parts to remove, hehe
The concern is whether it will affect the readability of these digits.

Pmant, 2023-01-13:
I just stopped the print before it got to the digits and counted the lines, they print after the PA-lines. This method works good for me and is much faster than the tower method.

psiberfunk, 2023-01-17:
> The digits are the most painful parts to remove, hehe The concern is whether it will affect the readability of these digits.

I'd take that risk for the lines  i think people will figure out which lines are which.. honestly as long as a few of the lines are readable the rest are easy to figure out :)

rlabusiness, 2023-01-24:
@SoftFever 
Thought I'd share some manually crafted proof-of-concept GCODE I've been using with good success over the last week or so, with digits remaining easy enough to read. I've included a screenshot of the expected result, as well as a video showing how easy it is to remove.

_NOTE: This sample GCODE is static, so it's only suitable for a printer with a 300x300 bed, such as the Creality CR-10 or its variants. Anyone who wants to try it, feel free, but use at your own risk._

[image]

VIDEO (showing ease of removal):
https://user-images.githubusercontent.com/6774974/214240935-a7aac9d0-c8e2-4d9a-b30b-7b903d503689.mp4

FILES:
[Sample GCODE to Append for Easy Removal of SoftFever PA Line Method Calibration 300x300 bed only.txt](https://github.com/SoftFever/BambuStudio-SoftFever/files/10487484/Sample.GCODE.to.Append.for.Easy.Removal.of.SoftFever.PA.Line.Method.Calibration.300x300.bed.only.txt)
The additional gcode for easy removal to copy and paste into a generated/saved calibration print file

[pressure_advance_test PLA 5m53s 1.0255g.gcode.zip](https://github.com/SoftFever/BambuStudio-SoftFever/files/10487494/pressure_advance_test.PLA.5m53s.1.0255g.gcode.zip)
The calibration print that I used in the video and screenshot after integrating the sample GCODE (zipped since Github doesn't accept .gcode files)

closed 2023-11-25 by github-actions[bot]