OrcaSlicer/OrcaSlicer #163: PA Tower Test : Fan And Layer Time Dont Match Up

psiberfunk, 2023-01-11:
**Describe the bug**
Something is wrong with either the layer time display or the fan computations under certain situations.    In this project file, I've got a PA test tower setup.. and I was noticing that although the fan speed should be 0 for any layer time >4 seconds. according to the filament settings,  the fan is running according to the "fan speed" view.

Therefore ... either the fan speed view is wrong, the layer times are wrong, or something else is getting bugged on when to run on the fan in the slicer code.

**3mf File for This Bug**
I have included a Zipped 3MF file here to replicate the issue: 
[PATest_FanAndLayerTimeDontMatch.3mf.zip](https://github.com/SoftFever/BambuStudio-SoftFever/files/10396759/PATest_FanAndLayerTimeDontMatch.3mf.zip)

**To Reproduce**
Steps to reproduce the behavior:
1. Open the 3MF file, click slice.
2. Inspect the layer times vs the fan times, note the inconsistency with the filament settings on when the fan should start to turn on (any time greater than 4 seconds should have no fan).

**Expected behavior**
The filament settings, fan speed, and layer time should all be self-consistent/in agreement.

**Screenshots**
These 3 screenshots show not all 3 sets of visualizations/settings are in agreement: 

[image]
[image]
[image]

**Desktop (please complete the following information):**
 - Windows or MacOS, does not matter.

SoftFever, 2023-01-12:
The fan turns off if I change the minimum layer time to 3.5 seconds.
I'm not entirely sure whether the different layer time calculation logic causes it. 
It is also possible that the auto fan speed logic works like that. 
I suspect the latter but I need to check ;)

psiberfunk, 2023-01-12:
> The fan turns off if I change the minimum layer time to 3.5 seconds. I'm not entirely sure whether the different layer time calculation logic causes it. It is also possible that the auto fan speed logic works like that. I suspect the latter but I need to check ;)

Yea, as you can see, something's getting bugged one way or the other.

Pmant, 2023-01-13:
Propably two different calculation for the layer time are done with the fan calculation being slightly faster. The fan speed calculation is based on this layer time calculation:
https://github.com/SoftFever/BambuStudio-SoftFever/blob/ff1303778dce64f4a8812f2de541f22d3213b88b/src/libslic3r/GCode/CoolingBuffer.cpp#L637

psiberfunk, 2023-01-13:
> Propably two different calculation for the layer time are done with the fan calculation being slightly faster. The fan speed calculation is based on this layer time calculation: [...]

Curious: is this disparity in upstream SuSi/PrusaSlicer or contained to within Bambu's fork ?

Pmant, 2023-01-13:
It's the same in Bambu Studio but they don't have the layer time in preview. I'm not sure which one is wrong.

psiberfunk, 2023-01-13:
> It's the same in Bambu Studio but they don't have the layer time in preview. I'm not sure which one is wrong.

yea, i'm wondering if it's ALSO present in superslicer/prusa

Pmant, 2023-01-13:
I wonder how much this issue actually matters since it does not seem to scale with layer time, seems more like a small offset per layer. The actual printed layer time will differ again due to motion planning done by the firmware where the fan speed won't be adjusted.

psiberfunk, 2023-01-13:
> I wonder how much this issue actually matters since it does not seem to scale with layer time, seems more like a small offset per layer. The actual printed layer time will differ again due to motion planning done by the firmware where the fan speed won't be adjusted.

It matters in as much as it's very confusing to try and figure out why a fan speed is not what you expect based on the layer time, which is a key reason to have the layer time view (to manage cooling issues).

Pmant, 2023-01-13:
I don't disagree. But the layer time is still only off by 0.x seconds. The issue only is as pronounced in this case because you set it to change 60% fan speed over 3s layer time. As mentioned, the slicer won't be able to tell the layer time this accurately anyway.

psiberfunk, 2023-01-13:
> I don't disagree. But the layer time is still only off by 0.x seconds. The issue only is as pronounced in this case because you set it to change 60% fan speed over 3s layer time. As mentioned, the slicer won't be able to tell the layer time this accurately anyway.

As it turns out this is actually a Bambu default for Nylon .. so it's not totally crazy:
[image]

closed 2023-11-25 by github-actions[bot]