OrcaSlicer/OrcaSlicer #166: Perimeters don't extrude enough to close without gap

npoulos, 2023-01-12:
**Describe the bug**
I posted this bug in original BS repo, but figured you guys might be able to address it in this fork.
( https://github.com/bambulab/BambuStudio/issues/1063 ) 

It appears that BS doesn't completely draw a perimeter which causes large gaps at the start/end of each line.

**3mf File for This Bug**
I've attached some gcode generated by Cura vs gcode generated by BS and you can see Cura almost completely closes the line whereas BS ends too soon. It's like a hard coded coasting. The coordinates draw a 1 layer, 1 line circle.

[testring-CURA-onelayer.txt](https://github.com/SoftFever/BambuStudio-SoftFever/files/10401523/testring-CURA-onelayer.txt)
[testring-BL-onelayer.txt](https://github.com/SoftFever/BambuStudio-SoftFever/files/10401524/testring-BL-onelayer.txt)

I've tested this using arcs and no arcs, and arachne on and off and still the same outcome.

**To Reproduce**
Slice any file and look at the start and end of a line. Turn off seam and retract/unretract and you'll see there's a clear gap between them that doesn't fill with filament (even wiping doesn't fix)

**Expected behavior**
A line should come to a complete close by default, or a "coasting" value should be allows to be added to control how big the gap at the end of a perimeter is

**Screenshots**
[image]
line start position

[image]
line end position (with gap shown) - Gap should be way smaller if there at all.

**Desktop (please complete the following information):**
 - OS: macOS
 - Version 1.4.2

SoftFever, 2023-01-12:
@npoulos this is an exciting finding. 
Have you compared it to PrusaSlicer by any chance? 
I suspect this is a different approach to handling seams implemented by Slic3r and Cura
I plan to examine this further over the weekend. We probably can introduce a new parameter for it.
more, well done on the discovery!

Naninani, 2023-01-12:
I reported this issue to BL back in Dec when I got my P1P and they suggested it could be related to retraction. However it is not. Disabling the wiping leaves a huge seam gap. And the retraction does not have a effect on the seam. 
We came to the conclusion that this might be related to LA/PA, but it very well may be pure software bug.

npoulos, 2023-01-12:
@Naninani this is 100% either a bug or a hard coded limitation. I've done a whole bunch of digging to see if anyone else has reported it. I've seen it brought up in Prusa forums from back in 2019 and the devs decided it was just the way the visual preview renderer works and wasn't a real problem and closed it off when really it was actually what's happening here.

To me it's pretty simple - if the perimeter is never drawn completely then theres no amount of LA/PA that will close off the gap between. The printhead just doesn't move over the gap while extruding so the seam can't be closed.

@SoftFever I then found that SuperSlicer actually caught the issue here: https://github.com/supermerill/SuperSlicer/issues/1534 and added a "seam gap" setting so the user can manually apply a seam gap. I've seen issues since though where people complain that the new feature doesn't work. https://github.com/supermerill/SuperSlicer/issues/2995

I'm not sure if the seam problem in SueprSlicer is the same but I think the "15% gap" that is mentioned in issue 1534 might have been added back in Slic3r days or something.

I did a little testing on Prusa slicer and found the issue is there yet slightly less pronounced. I'm going to try find some time to do further tests.

npoulos, 2023-01-13:
[image]

This line in src/libslic3r.h looks like it's causing the 15% gap. I have no idea if it is but it looks like it's the line that SuperSlicer changed to have a user entered number instead

npoulos, 2023-01-13:
I changed that number from 0.15 to 0.05 and compiled and while the gap in the viewer was smaller, prints had basically the same gap. When i changed it to 0 the gap on the print was much smaller but still there. Though at this point playhing with the wipe does manage to clean over it.

I do have a theory that it seems like if the last segment of the line the closes the gap is really small it's like it doesn't actually extrude the line which is where i see the gaps.

Is there potentially something defined that has a minimum e value?

SoftFever, 2023-01-14:
> 
E value should be the same as the perimeter.
It's likely the PA effect that kicked in near the gap.
I'm thinking about this behavior. 
Reducing the gap alone is not enough to improve the seam much. This guy's test is interesting, too: https://github.com/supermerill/SuperSlicer/issues/2995#issuecomment-1201404381

I think it is worth implementing something like `outer wall wipe distance` in Cura to improve the seam.

Naninani, 2023-01-14:
I think the gap we see is intentional. As when you put in perspective the Unretract length, wipe, LA/PA + print speed - all this contributes to closing it up.
I did some testing with Prusament PETG with disabled "Retract Amount before wipe"
[image]
and the major factor is the LA/PA.

npoulos, 2023-01-14:
Lowering to 0 definitely makes the gap slightly smaller, though doesn't close it completely.

I did have an issue i noticed after compiling, if i try to slice a bunch of items, specifically ones with 1 wall width lines it causes it to crash. Only happens when i changed the value from 0.15 to 0 - strange!

I did notice though, that if you do have it set to 0, and turn on wipe with retract, bring the retraction way low to 0.05 and add 99% retract before wipe with a wipe of like 10mm, it almost completely wipes the gap closed. (you need 99% retract before because 100% wont allow the wipe to happen)

I'm starting to think there are a couple things going on - due to the extremely high acceleration and speeds, the nozzle is actually sucking back up the deposited last dot of filament when it retracts, on top of this, when the nozzle moves away to do the next perimeter its ripping the last bit of filament away. 

By adding a 10mm wipe with virtually no retraction, the last dot of filament doesn't get sucked back up, or ripped away.

I actually think i even found some comments on this in the code in GCode.cpp - 
[image]

I tested slowing a print way way way down and the seam basically disapeared.

I'm thinking controls for the seam gap and the outer wall wipe distance (separate from retraction settings) that you mentioned @SoftFever would allow some really good tweaking of the settings to get virtually no seam.

I think a setting for the speed of the outer wall wipe distance, as well as allowing for it on the final z layer (to avoid the little string dot when a print finishes) would be helpful too.

SoftFever, 2023-01-14:
@npoulos oh, yeah.
That reminds me that I noticed the wipe speed issue and had some changes for my PrusaSlicer fork last year. I I implemented a `role based wipe speed` back to the time: https://github.com/prusa3d/PrusaSlicer/commit/4b3e0f980e29cc18813406e59fa7a1af68fd7bb0 ;)
I should port that change to this fork as well.

npoulos, 2023-01-14:
awesome! Could you actually point me in the direction of a commit you've made where you've added a feature with UI changes? I've attempted but can't seem to figure out how to add stuff to the UI to then pass data to the code. I'm an ex web-dev with experience in ColdFusion and a bunch of front end web languages but never used C++. Playing with this code has given me a little spark back haha. Would love to reverse engineer one of your commits to understand the UI side. Also what would you suggest as a good code editor? I'm just using Brackets. Would love to try and contribute!

SoftFever, 2023-01-14:
I've just pushed a commit that implements this seam_gap. 
This pretty much demonstrates how to add a parameter. You can take a look.

SoftFever, 2023-01-14:
NOTE: If anyone is interested in experimenting with this new parameter, you can access the latest CICD build on GitHub: https://github.com/SoftFever/BambuStudio-SoftFever/actions/runs/3918338007.
I designed this parameter in a way that it can be adjusted on a per-object basis, which should make it easier to find the optimal setting. 
I haven't printed with these changes by myself yet. 
Additional volunteers to help improve the seam are welcome. 
If all goes well, I'll add a new calibration method for the z seam.

[image]

npoulos, 2023-01-14:
@SoftFever Genius! Will do a bunch of testing tomorrow 😊

npoulos, 2023-01-16:
Been testing this a whole lot and while it doesn’t seem to make *much* of a difference at top speed, when you slow down acceleration you can really see the benefit of controlling this.

it seems the super high travel acceleration rips away material as it moves away so when you slow it down, closing off this gap works really well. 

Have had a couple crashes randomly but not
sure if it’s related to his enhancement. So far working really well.

ctadlock, 2023-01-27:
I have this issue and would love a solution. I got the build with this changed and tested it out. I see very little effect.  Thoughts?

Note that I had previously triend adjusting retraction speed and distance and it had very little effect if any.

[image]

npoulos, 2023-01-27:
@ctadlock slow down your travel acceleration by like half as well as travel speed to like 200-300. I found it helps a lot combined with the gap seam set to 0

ctadlock, 2023-01-27:
> @ctadlock slow down your travel acceleration by like half as well as travel speed to like 200-300. I found it helps a lot combined with the gap seam set to 0

That was printed with..

Outer wall speed: 60 mm/s
Outer wall acceleration: 2000 mm/s
Polylite PLA Pro
Retraction length: 0.8 mm
Retraction speed: 30 mm/s

ctadlock, 2023-01-27:
If you look closely along the Z axis its not just a gap in the seam. It seems like two issues:

- One of the ends of the outer wall path is pushed out
- The other end of the outer wall path ispushed in (has gap, dimple...)

[image]

npoulos, 2023-01-27:
I found the same issue. I'm almost certain it's related to the acceleration of travel moves - they pull the last bit of filament with the nozzle in whichever direction it moves.

I noticed in your settings you didn't mention the travel acceleration speed. This is the most important factor I've found. Here are my settings - They've been pretty successful in getting the seam as small as possible.

[image]

[image]

Mix that with a seam gap of 0 and see how it goes.

ctadlock, 2023-01-27:
Here are my settings, Ill try matching yours

[image]

[image]

[image]

ctadlock, 2023-01-27:
I tried all of your settings; the result looked different and worse and did not improve the problem. I tried just setting the travel speed and accel and it also did not make a significant effect.

Looking close at the actual path in gcode, the part of mine that is "out" of the circle and has a bulge is the last part of the outer wall path, and the part that is "in" the circle and has a gap is the first part of the outer wall path; then there is a wipe and travel to start the infil. The only travel I see that could have an effect is the travel to start the outer wall. You think that is happening too fast and pulling the material into the start of the outer wall?

[image]

--

Im also comparing against PrusaSlicer and my Mini+ which had no issues with seams at all. One thing I have noticed is that in Prusa once the main part of the outer wall is done there is a slight travel inwards and then back out to start the wipe. This does not happen in Bambu. Would this not have the effect of filling in that gap? What setting effects this?

Is ther a way to view the gcode in preview in Bambu like Prusa?

[image]

[image]

closed 2023-05-06 by SoftFever

grrminator, 2024-11-22:
This is still an ongoing problem. And I have tested seam gat with 0, 2.5%, 5%, 7.5% and 10%.