OrcaSlicer/OrcaSlicer #169: Security: .DS_Store files in the Mac app

Naninani, 2023-01-12:
**Describe the bug**
.DS_Store files are included in many dir when the .app is packaged. This is a security issue as .DS_Store files stores custom attributes/metadata of its containing folder and the names of other files around it. Exposing this information could potentially allow hackers to act maliciously and let them see private files.

**To Reproduce**
I used Beyond Compare to compare the both the official Bambu app and SoftFeaver's one.

**Expected behavior**
.DS_Store should be included in the .app.
More: https://buildthis.com/ds_store-files-and-why-you-should-know-about-them/

**Screenshots**
Filtered to show only the added .DS_Store files. 
[image]

**Desktop (please complete the following information):**
 - OS: Mac 
 - Version [V1.4.2](https://github.com/SoftFever/BambuStudio-SoftFever/releases/tag/v1.4.2)

SoftFever, 2023-01-13:
Thanks for the heads up.
It looks like they exist on the git repo. Will delete them

SoftFever, 2023-01-14:
This change dd1efb3969093bd44fe070644e18018d3f517213 should fix it.
Thanks again.
I'm closing this issue now.

closed 2023-01-14 by SoftFever