OrcaSlicer/OrcaSlicer #170: output wipe tower position x y in GCODE more then once

hliebscher, 2023-01-13:
in G-Code file an the end isprinter config
it is wrong output for 7 build plates but i use only one and size is 200/200mm
; wipe_tower_x = 155,155,155,155,155,155,155
; wipe_tower_y = 165,240,240,240,240,240,240

i need this format for post processing palette 3 pro 
; wipe_tower_x = 155
; wipe_tower_y = 165

hliebscher, 2023-01-13:
is it possible to add prusaslicer like output? for Pint bed size?

bed_shape = 0x0,278x0,278x245,0x245 

and bambustudio

printable_area = 0x0,278x0,278x245,0x245

hliebscher, 2023-01-13:
[Box-1x1.3mf.txt](https://github.com/SoftFever/BambuStudio-SoftFever/files/10413438/Box-1x1.3mf.txt)

wipe_tower_x     value is random how often is generated (in this sample 4 times)
 its depends also from slice all or slice plate

hliebscher, 2023-01-13:
[project_settings.config.txt](https://github.com/SoftFever/BambuStudio-SoftFever/files/10413884/project_settings.config.txt)

i found this it is in old 3mf files in project_settings.config

    "wipe_tower_no_sparse_layers": "0",
    "wipe_tower_rotation_angle": "0",
    "wipe_tower_x": [
        "84.8135",
        "133.572"  <<<<<<<<<<<<<<-------------false
    ],
    "wipe_tower_y": [
        "147.728",
        "118.349"  <<<<<<<<<<<<<<-------------false

SoftFever, 2023-01-18:
Fixed.
You can grab the dev build here to give it a try: https://github.com/SoftFever/BambuStudio-SoftFever/actions/runs/3950164448

hliebscher, 2023-01-22:
very nice it works !

closed 2023-01-23 by hliebscher