OrcaSlicer/OrcaSlicer #174: Feature Request to add PrusaSlicer/SuperSlicer 'Label Objects'

eriroh, 2023-01-13:
Would be superb if the 'Label Objects' functionality could be added, so objects can be excluded through mainsail/fluidd/etc. 

[image]

closed 2023-01-13 by eriroh

sickerin, 2023-02-04:
Was this added, hence closed? If not I think it would be good to have. Also from what I recall, the Bambu Studio team does have plans to add options to exclude objects mid print.

eriroh, 2023-02-04:
I closed this because BambuSoftFever labels objects by default (not sure about the official bambu studio as I've never used it). I just didn't realize that I had my [exclude_objects] in the wrong config.