OrcaSlicer/OrcaSlicer #175: Switch Timelapse crash BambuStudio

vr6syncro, 2023-01-13:
**Describe the bug**
BambuStudio Crash on switch Timelapse

**To Reproduce**
Steps to reproduce the behavior:
Open BambuStudio and Switch between Traditionell to Regluar

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Win 11
 - Version latest

[Screenshot 2023-01-13 235556](https://user-images.githubusercontent.com/27892314/212433969-e809ccd8-1066-49e2-bcc7-96f54dd80f92.png)

SoftFever, 2023-01-14:
@vr6syncro this bug should have been fixed in 1.4.2.
Can you check what version are you running?

vr6syncro, 2023-01-14:
[image]

It is the 1.4.2

SoftFever, 2023-01-15:
A sample project that can repro this issue will be helpful.

vr6syncro, 2023-01-16:
U can watch a video from the Error: https://www.youtube.com/watch?v=M-NdgMnwUMI
No Project needed.
[log.txt](https://github.com/SoftFever/BambuStudio-SoftFever/files/10428329/log.txt)

SoftFever, 2023-01-17:
@vr6syncro 
Hmm, I'm not able to repro it.
Can someone help to verify this?

vr6syncro, 2023-01-17:
> @vr6syncro Hmm, I'm not able to repro it. Can someone help to verify this?

is there anything i can do to get more infos about the crash that might help?

MaxKW1, 2023-01-22:
I too get this on the latest release here, as well as the normal BStudio with a M1 Mac

MaxKW1, 2023-01-22:
Here is my project, for me just opening it and then switching the timelapse option to smooth with result in the app closing out. 
[App Crash When Switching to Smooth Timelapse.3mf.zip](https://github.com/SoftFever/BambuStudio-SoftFever/files/10474980/App.Crash.When.Switching.to.Smooth.Timelapse.3mf.zip)

kevinpoitra, 2023-01-23:
I can reproduce this 100% of the time. For me, it doesn't matter what object I'm printing - I can submit one print and have it switch to the video feed screen without issue, but upon printing any second object (regardless of it being the same object or different), Bambu Studio will crash the moment the timer to switch to the video feed hits 0.

I'm on Pop!_OS 22.04 running version 1.4.2.

SoftFever, 2023-01-24:
Thanks.
I'm able to reproduce the crash with attached sample project.
PR #224 should fix this issue. 
If you want to help verify the fix, you can download the dev build to try the fix: https://github.com/SoftFever/BambuStudio-SoftFever/actions

vr6syncro, 2023-01-27:
The latest solve the problem, thank you.

closed 2023-01-27 by vr6syncro