OrcaSlicer/OrcaSlicer #180: Default settings for Different Bed Types

IzzyMan12078, 2023-01-15:
Quite often I forget to disable Flow Calibration when sending prints to my printer while using the textured pei plate from Bambu.  Many times this calibration causes issues for me.

It would be great if by default, that option was not checked when using the textured PEI or other plates were flow calibration can not read well..  I suppose ideally, a settings page for plates would be great.  The slicer already disables first layer calibration/lidar when using the textured pei plate.

IzzyMan12078, 2023-01-15:
Very similar to $177 - but want retained per bed type if possible.  I will never want Flow Calibration defaulted to checked with Textured PEI however I will almost always want it defaulted to checked with cool plate or others? - Also - can we adda bed type?

SoftFever, 2023-01-16:
This logic will conflict with #177 

To keep things simple, the application will remember these options.