OrcaSlicer/OrcaSlicer #186: Extend Maximum Size Limitation

madeintaiwan, 2023-01-17:
Currently, the bed limitation is 1200mm and our printer is 4000mm long. 
Is it possible to either remove the maximum value limit or set it at something like 6000mm?

SoftFever, 2023-01-17:
That's huge!
Will ease the limitation in the next update.
Would love to see how does it work on such large printer :)

madeintaiwan, 2023-01-17:
Pretty big, we're building a 6000mm3 one currently so we can print a small house, Superslicer has similarly removed the upper threshold to allow us to do some pretty big prints- 
[image]

madeintaiwan, 2023-02-23:
Hi, any chance you had a look at removing / changing the bed size limit?

Thanks,

Nate

madeintaiwan, 2023-11-17:
> That's huge! Will ease the limitation in the next update. Would love to see how does it work on such large printer :)

@Softfever any chance you can increase the bed size limit?

SoftFever, 2023-11-18:
I'll take another look for 1.9.
It needs to change change to internal data type to support larger bed size

madeintaiwan, 2023-11-18:
Thanks, that would be great. 

There seems to be a 2500mm limit in Y axis and a 2000mm limit in Z and I assume a similar limit in X but I haven't tried. 

Check out tldraw (make it real) and their Twitter feed. Possibly ready for Orca 2.0....

madeintaiwan, 2024-02-09:
@SoftFever Any chance there's room for this in the next update? 🙏🙏 Having to go back to Cura to print big :( 

[image]

madeintaiwan, 2024-05-10:
@SoftFever Any chance you had a look at why there's a limit for length and if it possible to extend for larger format printers ?. 

Thanks , Nate

SoftFever, 2024-05-10:
> [@SoftFever](https://github.com/SoftFever) Any chance you had a look at why there's a limit for length and if it possible to extend for larger format printers ?.
> 
> Thanks , Nate

working on it now

PR #5320 (2024-05-17): Support larger printer (up to hundred meters)

# Description

Orca Slicer currently only supports printer bed sizes smaller than 2147mm x 2147mm due to the limitations of the 32-bit data type.
This limitation also applies to the maximum number of platers you can create for a project.
The total width/length of all the plates cannot exceed this limit.

With this change, Orca can now support sizes as large as desired.
For convenience, the limit is currently set to 214m x 214m x 214m.

SuperSlicer is referenced for some changes.

closes #186 
# Screenshots/Recordings/Graphs

[image]

- [x] Fix the [... cut]
closed 2024-05-11 by madeintaiwan