OrcaSlicer/OrcaSlicer #187: Icon switches colors on MacOS  (cosmetic bug)

psiberfunk, 2023-01-17:
**Describe the bug**
On macos, during application launch the app icon in the Dock has the correct new color scheme (teal), but switches to the old green color after launching.

During launch :  
[image]

After Launch, you see this:
[image]

Mac OS only, as far as I can tell.  Running intel/x86 version.

hliebscher, 2023-01-17:
i am looking also for button slice plate and sen g-code ;-)

SoftFever, 2023-01-17:
Has the fix ready locally :)
Will push the fix later.

hliebscher, 2023-01-17:
ok

psiberfunk, 2023-01-17:
> ok

you seem to be commenting on the wrong issue.. perhaps thats why you're confused?

IkarusX1978, 2023-01-17:
same issue here ...
concerns even Intel and Sillicon Mac!

hliebscher, 2023-01-18:
> > ok
> 
> you seem to be commenting on the wrong issue.. perhaps thats why you're confused?
i had fixt this with pa but to late  SoftFever has fix locally  ...

hliebscher, 2023-01-22:
@psiberfunk can you close ? its fixed

closed 2023-01-22 by psiberfunk