OrcaSlicer/OrcaSlicer #192: Calibration wiki page to disable Flow Calibration

fleutot, 2023-01-18:
I made the Flow Calibration thing according to the wiki, twice. After the second time, I noticed the note "If you are using an X1 or X1C printer, make sure you do not select the 'Flow calibration' option.".

So now I need to do it a third time.

Suggestion: move that "Note" to the top of the page. There are two occurrences of the note, so either mention that it needs to be off for all operations on the page, or move it twice to the top of their respective sections.

Ixian, 2023-01-20:
If you calibrate manually and are happy with it you want to leave Flow Calibration off. 

Currently you need to toggle it off for every print, which is annoying - I think there is a feature request in with BL Labs to enable persistently toggling it off (like you can now do with the plate checking, for example) but you might want to check and/add your support to it, they have a ton of little requests in the queue.

SoftFever, 2023-01-22:
The  Calibration Wiki is meant to be for all printers.
Hence the note part is not at the top of the page.
Thanks for the suggestion anyway.

closed 2023-01-22 by SoftFever