OrcaSlicer/OrcaSlicer #199: Big Fan is not enabled after layer 1

VexAzomi, 2023-01-19:
As the title says, I noticed, that the big cooling fan (P2) is not enabled after layer 1.
I traced down the bug to be a missing line in the gcode.
As far as I am aware, the only setting for the big fan (P2) can be made in the filament cooling section.
If I am wrong, please correct me. However here are the relevant lines of two gcodes:

This is the official slicer:
M106 S252.45
M106 P2 S204
; CHANGE_LAYER
; Z_HEIGHT: 0.4
; LAYER_HEIGHT: 0.2
G1 E-.56 F1800
; WIPE_START
G1 F24000
........

This is from softfever:
M106 S252.45 ; enable fan
; CHANGE_LAYER
; Z_HEIGHT: 0.4
; LAYER_HEIGHT: 0.2
G1 E-.56 F1800 ; retract
; WIPE_START
G1 F24000
........

hliebscher, 2023-01-19:
witch printer ? have you enable aux fan in printer Config?

tripp396, 2023-01-19:
I am seeing this on my P1P and newly installed and confirmed working fan. Definitely goes for the first layer then shuts off. Filament cooling on that profile has aux fan set to 70%.

VexAzomi, 2023-01-19:
Dammit, I am sorry. Tried to note everything important but my brain is stupid.
Okay, so I am printing on the P1P with the textured sheet. I made a copy of the BBL PLA filament config and the 0.2mm standard settings. What Printer config are you referring to?

tripp396, 2023-01-19:
For me it is the following profiles, all stock:

Printer: Bambu Lab P1P 0.4 nozzle
Filament: Bambu PLA Basic
Process: 0.16 Optimal @BBL P1P (only change here is supports added and a different top pattern)

VexAzomi, 2023-01-19:
tripp396, can you explain what you are experiencing? I dont really get what you mean by "Definitely goes for the first layer then shuts off"

tripp396, 2023-01-19:
> 

Sure. The auxiliary fan will blow air for the first layer of the print. Then all subsequent layers the auxiliary fan will not run at all.

VexAzomi, 2023-01-19:
Äh, did you turn off the function disable fans for the first layer?
Because... the fan is supposed to do the opposite or not?

tripp396, 2023-01-19:
I did not enable that. I could enable and see if its working in reverse. I'll have to see what my schedule looks like today for that test.

I'm not sure its that because the only fan not working is the big part cooling fan (the free gift one). Not the part cooling one on the print head.

VexAzomi, 2023-01-19:
So uhm I found the setting panel hliebscher was referring to. So many new buttons with this program.. I did not notice, that there was a button 
[image]
After enabling that, everything works now, so I am going to close this issue.

closed 2023-01-19 by VexAzomi

tripp396, 2023-01-19:
> So uhm I found the setting panel hliebscher was referring to. So many new buttons with this program.. I did not notice, that there was a button [image] After enabling that, everything works now, so I am going to close this issue.

Well I feel like an idiot. I thought I looked through there like 5 times.

VexAzomi, 2023-01-19:
Yep, I feel like an idiot for opening an issue here. But hey, if its working for you aswell... hurray!

CCS86, 2023-02-27:
I still think there is an issue here with the default printer profile not having the AUX fan enabled. You can save a user preset of the machine with it turned on, but you really shouldn't have to.