OrcaSlicer/OrcaSlicer #204: Segfault Linux AppImage on Login

urbanslayer, 2023-01-21:
I haven't seen this listed as a known issue (might have missed it) or an open/closed issue.  On Arch, the slicer works fine unless I login/link my account (which is a Google account).  Once done, the app freezes and then closes.  Any further attempt to reopen it also results in an immediate close.  If you delete the ~/.config/BambuStudio-SoftFever directory, the app will load again until you login again.

On running this from the command line, on one system I get nothing, on another just a single line indicating a segfault - 

_segmentation fault (core dumped) ./BambuStudio_ubu64.AppImage_

From the journal (error is the same every time) - 

_systemd-coredump[263969]: [🡕] Process 263817 (bambustu_main) of user 1000 dumped core.
                                                
                                                Stack trace of thread 263817:
                                                #0  0x00007f4c4ecafb4e __libc_free (libc.so.6 + 0x96b4e)
                                                #1  0x00007f4c41172d58 ERR_clear_error (libcrypto.so.3 + 0x172d58)
                                                #2  0x00007f4c2ba9e6cf n/a (/home/lain/.config/BambuStudio-SoftFever/plugins/libbambu_networking.so + 0x29e6cf)
                                                ELF object binary architecture: AMD x86-64_

Log with stamp - 21_12_02_16 is from opening and then trying to login.
Log with stamp - 21_12_04_10 is trying to open after initial crash after logging in.

[debug_Sat_Jan_21_12_02_16_240615.log](https://github.com/SoftFever/BambuStudio-SoftFever/files/10472128/debug_Sat_Jan_21_12_02_16_240615.log)
[debug_Sat_Jan_21_12_04_10_245268.log](https://github.com/SoftFever/BambuStudio-SoftFever/files/10472129/debug_Sat_Jan_21_12_04_10_245268.log)

MissingNoIOI, 2023-10-26:
Yeah, I have the same problem on Archlinux. Tried both the version from the AUR and the AppImage directly and I cant log in with either. First time logging in I'm getting an error on the website that something went wrong and the second time OrcaSlicer segfaults

Jo-Con-El, 2023-12-05:
@urbanslayer, @MissingNoIOI: Have you tried 1.8.0 or 1.8.1?  There are other reports of people running Orca successfully.

urbanslayer, 2023-12-05:
> @urbanslayer, @MissingNoIOI: Have you tried 1.8.0 or 1.8.1? There are other reports of people running Orca successfully.

Yes, it appears to be working for me.  I haven't had a chance to print yet, but signing in etc all works and the app does not crash.  Thanks!

AdamLeyshon, 2024-02-27:
Building orca-slicer-1.9.0-1 from AUR still has the same problem

AlmightyFrog, 2024-04-28:
Same issue here using OrcaSlicer from AUR (still on 1.9.1-1 as 2.0.0-1 does not install).

At least deleting `~/.config/OrcaSlicer/BambuNetworkEngine.conf` does allow me to use the slicer again.

All I want is sync my printer and filament profiles - is there a other way? I do not want to go the manual git repo sync way as i am too lazy to keep it up-to-date.

closed 2024-08-05 by github-actions[bot]