OrcaSlicer/OrcaSlicer #208: no Bed Type  to choice.

goldjunge91, 2023-01-21:
Hi,
in the official BambuStudio version I can choice between, all Bambu build plates, but on your version not, are this not implemented on your version ?
I added my anycubic i3 mega s profile to your anycubic max profile and added it as zip file.
and any idea how I can add plates ? and add to different plates custom settings would be great.
I'm not sure that its a Bug or Feature request.

[image]
[image]
[SoftFever.zip](https://github.com/SoftFever/BambuStudio-SoftFever/files/10473186/SoftFever.zip)

SoftFever, 2023-01-22:
It is per design.
As far as I know, no third-party printers use different types of beds.
They also tend to introduce more user errors so I disabled it.

closed 2023-01-22 by SoftFever

goldjunge91, 2023-01-22:
alright how can I enable it ? where I have to search in the code for it. I have differents plate and love to use the function

SoftFever, 2023-01-22:
I can’t remember the exact place. You can search through the commit history with keyword ’bed’Sent from my iPhone