OrcaSlicer/OrcaSlicer #213: Support Video Feed on Linux for BambuLab Printer

SebiPanther, 2023-01-22:
**Is your feature request related to a problem? Please describe.**
If you try to start the video feed of a BambuLab Printer on Linux this error message appear:
[image]

**Describe the solution you'd like**
This solves it: https://github.com/bambulab/BambuStudio/pull/1045
Maybe if you update your fork and do a new release its already fixed.

**Describe alternatives you've considered**
Update the fork from this source directly: https://github.com/jwise/BambuStudio/tree/jwise/x1-video-linux

**Additional context**
-     OS: Ubuntu Mate 22.10
-     Version V1.4.2

Thanks for the good work!

SoftFever, 2023-01-22:
It's merged.
You can test it with the nightly build here: https://github.com/SoftFever/BambuStudio-SoftFever/actions/runs/3977966683

SebiPanther, 2023-01-22:
Yes - it works - thank you! \o/
[image]
Sidenote: I had to install a something, but the Warned me about it:
[image]
I just did this and it worked well: `sudo apt-get install gstreamer1.0-plugins-bad gstreamer1.0-plugins-bad-apps`

closed 2023-01-22 by SebiPanther