OrcaSlicer/OrcaSlicer #217: Unable to set layer height to .3 using .4 nozzle

feistyknitty, 2023-01-22:
**Describe the bug**
Found in 1.4.2: 

There are designs (specifically "string art" designs from 3DPrintbunny) which require a layer height of .3. I have been able to use this layer height with .4 nozzle with no issues on multiple printers, including the BBL X1. Bambu Studio from BBL allows me to set .3 layer height with no issue. 

SoftFever branch gives me an error, and automatically resets layer height to .28. It does not allow me to override this. As this does not seem to be something that could cause harm to hardware (maybe just produce prints that aren't great quality if someone tries it without adjusting settings), I feel the user should be allowed to override this if they desire. 

Many sources on a Google search indicate that .32 is an acceptable max layer height for a .4 nozzle (e.g. https://3dsolved.com/best-layer-height-for-3d-printing/). I understand that with the speed of the Bambu, having a thicker layer height could cause issues with volumetric speed. If that is a concern, perhaps displaying an alert to the user about potential issues would be advisable, but preventing users from using a setting that **should** be usable is a poor user experience.

**3mf File for This Bug**
N/A

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Quality'
2. Click on 'Layer height'
3. Change field contents to ".3"
4. See error

**Expected behavior**
Layer heights up to potentially .32 should be allowed. For BBL printers, perhaps an alert that a high layer height could cause issues at high speeds could be helpful, but user should not be prevented from setting a layer height that is acceptable for printing with the selected nozzle.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: MacOS Monterey 
 - Version 12.4

hliebscher, 2023-01-22:
this is in your pinter settings under extruder!  pls change max from 0.28 to 0.32 than it works 
is not a bug ! its a feature

SoftFever, 2023-01-23:
@feistyknitty 
As @hliebscher mentioned, you can change the max layer height in extruder tab
[image]

feistyknitty, 2023-01-23:
Ah, thank you! I didn't realize this was a setting that could be changed. When I got the error on SoftFever and not main Bambu, I thought it was something specific to the branch. I do still find it odd that the default between the two is different, though!

closed 2023-01-23 by feistyknitty