OrcaSlicer/OrcaSlicer #219: Error running JavaScript message when launching BambuStudio

ChaosBlades, 2023-01-23:
[image]

Also seeing a lot of Critical/Warning messages.
https://gist.github.com/ChaosBlades/ee8bbc64ca88a61debdbcd80d2358fdd

I'm running Pop!_OS 22.04 LTS x86_64

ChaosBlades, 2023-04-17:
Issue was mentioned when the appimage builder was merged.
https://github.com/SoftFever/OrcaSlicer/pull/86

ckvsoft, 2023-04-17:
Issue persists also on beta OrcaSlicer v1.6.2
Ubuntu

xpsKING, 2023-04-25:
Same issue here on Nobara 37 (Fedora). Also turns the entire / drive read only until restart

ckvsoft, 2023-04-30:
Issue persists also on non beta OrcaSlicer v1.6.2
Ubuntu

Poikilos, 2023-06-01:
Confirmed in 1.6.3 final on Fedora 38

ChaosBlades, 2023-06-01:
I also still get it on the same version but on pop os.

byteofsoren, 2023-06-08:
Confirmed on ArchLinux.

```
10:48:21 PM: Error running JavaScript: Unsupported result type
10:48:21 PM: Error running JavaScript: Unsupported result type
```

Flavius-Geani, 2023-06-14:
Confirmed on ArchLinux and gtk seems to be mad too, not sure if there is a relation.
`(orca-slicer-bin:183254): Gtk-CRITICAL **: 16:57:16.163: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer-bin:183254): Gtk-CRITICAL **: 16:57:16.168: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

..........
`

rajinder-yadav, 2023-06-30:
## Version: V1.6.3

## System Info

Operating System: openSUSE Tumbleweed 20230628
KDE Plasma Version: 5.27.6
KDE Frameworks Version: 5.107.0
Qt Version: 5.15.10
Kernel Version: 6.3.9-1-default (64-bit)
Graphics Platform: X11
Processors: 24 × AMD Ryzen 9 5900X 12-Core Processor
Memory: 62.7 GiB of RAM
Graphics Processor: NVIDIA GeForce RTX 3080/PCIe/SSE2
Manufacturer: Micro-Star International Co., Ltd.
Product Name: MS-7A38
System Version: 8.0

## JavaScript error message

[image]

## Terminal output

```sh
$ ./OrcaSlicer_ubu64.AppImage 
[2023-06-30 11:55:03.625702] [0x00007f43de37ca40] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
loading network module at /home/yadav/.config/OrcaSlicer/plugins/libbambu_networking.so
error, dlerror is /home/yadav/.config/OrcaSlicer/plugins/libbambu_networking.so: cannot open shared object file: No such file or directory
after dlopen, network_module is (nil)

(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.368: gtk_window_resize: assertion 'height > 0' failed
(orca-slicer:16165): GLib-GObject-CRITICAL **: 11:55:08.378: invalid cast from 'wxPizza' to 'GtkCellLayout'
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.378: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
(orca-slicer:16165): GLib-GObject-CRITICAL **: 11:55:08.379: invalid cast from 'wxPizza' to 'GtkCellLayout'
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.379: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
(orca-slicer:16165): GLib-GObject-CRITICAL **: 11:55:08.380: invalid cast from 'wxPizza' to 'GtkCellLayout'
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.380: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
(orca-slicer:16165): GLib-GObject-CRITICAL **: 11:55:08.380: invalid cast from 'wxPizza' to 'GtkCellLayout'
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.381: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
(orca-slicer:16165): GLib-GObject-CRITICAL **: 11:55:08.385: invalid cast from 'wxPizza' to 'GtkCellLayout'
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.385: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
(orca-slicer:16165): GLib-GObject-CRITICAL **: 11:55:08.386: invalid cast from 'wxPizza' to 'GtkCellLayout'
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.386: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.416: gtk_window_resize: assertion 'height > 0' failed
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.418: gtk_widget_set_size_request: assertion 'width >= -1' failed
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.418: gtk_widget_set_size_request: assertion 'width >= -1' failed
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.419: gtk_widget_set_size_request: assertion 'width >= -1' failed
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.419: gtk_widget_set_size_request: assertion 'width >= -1' failed
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.419: gtk_widget_set_size_request: assertion 'width >= -1' failed
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.419: gtk_widget_set_size_request: assertion 'width >= -1' failed
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.419: gtk_widget_set_size_request: assertion 'width >= -1' failed
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.419: gtk_widget_set_size_request: assertion 'width >= -1' failed
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.426: gtk_window_resize: assertion 'height > 0' failed
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.426: gtk_widget_set_size_request: assertion 'width >= -1' failed
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.426: gtk_widget_set_size_request: assertion 'width >= -1' failed
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.494: gtk_window_resize: assertion 'height > 0' failed
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.499: gtk_widget_set_size_request: assertion 'width >= -1' failed
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.499: gtk_widget_set_size_request: assertion 'width >= -1' failed
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.506: gtk_widget_set_size_request: assertion 'width >= -1' failed
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.506: gtk_widget_set_size_request: assertion 'width >= -1' failed
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.512: gtk_widget_set_size_request: assertion 'width >= -1' failed
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.512: gtk_widget_set_size_request: assertion 'width >= -1' failed
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.517: gtk_widget_set_size_request: assertion 'width >= -1' failed
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.517: gtk_widget_set_size_request: assertion 'width >= -1' failed
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.523: gtk_widget_set_size_request: assertion 'width >= -1' failed
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.523: gtk_widget_set_size_request: assertion 'width >= -1' failed
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.529: gtk_widget_set_size_request: assertion 'width >= -1' failed
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.529: gtk_widget_set_size_request: assertion 'width >= -1' failed
[... 26 lines cut ...]
(orca-slicer:16165): GLib-GObject-CRITICAL **: 11:55:08.743: invalid cast from 'wxPizza' to 'GtkCellLayout'
[... cut ...]
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.827: gtk_widget_set_size_request: assertion 'width >= -1' failed
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.827: gtk_widget_set_size_request: assertion 'width >= -1' failed
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.834: gtk_window_resize: assertion 'height > 0' failed
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.835: gtk_widget_set_size_request: assertion 'width >= -1' failed
(orca-slicer:16165): Gtk-CRITICAL **: 11:55:08.835: gtk_widget_set_size_request: assertion 'width >= -1' failed
```

nettucu, 2023-07-12:
Confirmed on Arch Linux, orca slicer 1.6.3;

Error running Javascript: Unsupported result type

psi-4ward, 2023-07-25:
Several month later and still using PrusaSlicer :\

GauthamYerroju, 2023-07-27:
Confirmed on Nobara 37, Gnome 43.2, X11 session.

Error running Javascript: Unsupported result type

kylek29, 2023-07-27:
> Several month later and still using PrusaSlicer :\

You can't just "okay" the error and use it? That's how I use it on Ubuntu. It's a nuisance, but I haven't found anything it actually breaks.

srsam89, 2023-08-01:
> > Several month later and still using PrusaSlicer :\
> 
> You can't just "okay" the error and use it? That's how I use it on Ubuntu. It's a nuisance, but I haven't found anything it actually breaks.

Not for me. I click ok, it opens the error in another window and then the application freezes for a minute before crashing.

xpsKING, 2023-08-01:
I’ve just given up on a fedora build. BambuStudio has merged in most of
orca’s changes and they have a fedora app image

mplogas, 2023-08-05:
Bambulabs latest release (Fedora build 1.07.02.51) has this JavaScript error fixed for me. OrcaSlicer 1.6.4b2 still shows the JS type error.

Manjaro Linux, KDE/Plasma

discip, 2023-08-24:
I am getting the exact same error here on Manjaro Gnome with the latest release **[1.6.6](https://github.com/SoftFever/OrcaSlicer/releases/tag/v1.6.6)**.

WZ9V, 2023-08-26:
Still getting this Orca 1.6.6 on Linux Mint 21.2

iroCheese, 2023-09-07:
Same problem on Arch Linux with orca-slicer-bin-1.6.4-1.

ChaosBlades, 2023-09-10:
Fixed or at least hidden on OrcaSlicer V1.7.0 Beta Release

closed 2023-09-10 by ChaosBlades