OrcaSlicer/OrcaSlicer #222: Cannot open video

smolz, 2023-01-24:
**Describe the bug**
Click on the play button in the device screen and get a popup.
"BambuSource has not correctly been registered for media playing! Press Yes to re-register it."

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:

Click on the play button in the device screen and get a popup.
"BambuSource has not correctly been registered for media playing! Press Yes to re-register it."

Click yes, retry stream - NO CHANGE
Click yes, close app, retry stream - NO CHANGE
Right click app, try stream, same popup, Click yes - NO CHANGE

**Expected behavior**
Click play button, see live video of printer

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
windows, 22h2

yojoshb, 2023-02-05:
I encountered the same issue on my end (Windows 10, 22H2). The workaround for me was installing BambuStudio from BambuLab alongside this fork. I only use SoftFever's version to actually slice/print, etc and the video stream works now. I'm guessing the BambuLab version installs some additional libraries or set's up the initial connection to their cloud in a different way. I don't really know but it has been working for me.

derpicknicker1, 2023-03-27:
This is caused by a missing registry key. You can add it manually:

```
HKEY_CLASSES_ROOT\bambu
    Source Filter    REG_SZ    {233E64FB-2041-4A6C-AFAB-FF9BCF83E7AA}
```

[image]

Icesythe7, 2023-04-07:
> This is caused by a missing registry key. You can add it manually:
> 
> ```
> HKEY_CLASSES_ROOT\bambu
>     Source Filter    REG_SZ    {233E64FB-2041-4A6C-AFAB-FF9BCF83E7AA}
> ```
> 
> [image]
> 
> @yojoshb Your workaround works because Bambu Studio creates this registry key.
> 
> See also: https://wiki.bambulab.com/en/software/bambu-studio/faq/live-view -> Problem 3

This fixed it for me, just created the key and then added the data posted and worked instantly! TY!

lauhayden, 2023-04-14:
For ease of use:
1. Open a new text file using Notepad
2. Paste the following text inside:
```
Windows Registry Editor Version 5.00 

[HKEY_CLASSES_ROOT\bambu]
"Source Filter"="{233E64FB-2041-4A6C-AFAB-FF9BCF83E7AA}"
```
3. Save as `orcastudio_video_fix.reg`
4. Double-click the file, click yes twice
5. ✨ Profit! ✨

closed 2023-11-25 by github-actions[bot]