OrcaSlicer/OrcaSlicer #225: Custom Gcode before each object/part in an object

npoulos, 2023-01-24:
**Is your feature request related to a problem? Please describe.**
Trying to create my own calibrations / test out some things and editing the Gcode manually after is tedious.

For example, editing the pressure advance setting to calibrate it.

It would be awesome to be able to easily edit *all* settings for individual objects but i imagine that would be super cumbersome to allow so the next best thing would be to be able to add custom gcode before that object prints. Even something simple as adding a comment in the gcode so you know when the object starts etc.

**Describe the solution you'd like**
Custom gcode before / after each object

**Describe alternatives you've considered**
More options that can be changed per object  (LA/PA, retraction settings, etc.) I know you could just make more materials but it becomes super hard to manage

**Additional context**
thanks 😊

cotepat, 2023-01-24:
>Even something simple as adding a comment in the gcode so you know when the object starts etc.

The objects are already labeled in the gcode (comment starts with "; printing object")

npoulos, 2023-01-25:
Oh I totally missed that! I’ll check it out. would still be great to not have to mess with the exported gcode and instead be able to add in during setup.

SoftFever, 2023-02-26:
While per-object settings could be useful in some cases, I think the scenario you mentioned is best suited to a tower benchmark model, like the PA tower?
In that case, we can use custom gcode for layer change.

closed 2023-11-25 by github-actions[bot]