OrcaSlicer/OrcaSlicer #227: AppImage build re-work

fang64, 2023-01-24:
@SoftFever I'm going to refactor the build so it can grab dependencies from the host OS building it that are missing from the AppImage so it's way more portable. 

SubSurface from Linux Torvalds build approach seems way more sane as he's leveraging linuxdeployqt to grab the dependencies out before building with AppImage tool. I'm going to avoid the build script because it seems flawed for the linux build. I want to resolve these segfault/hard crash login issues and they don't happen if the libraries are grabbed from the OS that built it. There is a limitation if you build on too new of a distribution it will segfault on the older systems. so I think 20.04 is a reasonable compromise to support most...

I suspect it'll fix #204 and #185 as the buildlinux.sh script seems like a bear to use to fix the dependency issues.

SoftFever, 2023-01-26:
Awesome!👍 
Looking forward to seeing your new work!

emelbardis, 2023-01-29:
I'm fighting with #185 and look forward to your build.

andypiper, 2023-02-14:
Not sure whether this is useful, but I'm most of the way there (I think?) with building from source on an RPM-based system (in my case Fedora 37) but I needed to install these dependencies:

```
dnf install gtk3-devel libsecret-devel gstreamer1-devel libmspack-devel webkit2gtk4.0-devel mesa-libOSMesa-devel libcurl-devel openssl-devel eglexternalplatform-devel extra-cmake-modules libgudev-devel dbus-develm4 glew-devel perl-core wayland-protocols-devel
```

Having said that, I'm currently stuck with part of the dependencies section of the build from source, it seems to be stumbling on something to do with Boost.

Did you get a source tree into a state where it is buildable on Linux?

fang64, 2023-02-15:
> Not sure whether this is useful, but I'm most of the way there (I think?) with building from source on an RPM-based system (in my case Fedora 37) but I needed to install these dependencies:
>
> ```
> dnf [...]

There are instructions on the official Bambu Studio repository to build [Linux-Compile-Guide](https://github.com/bambulab/BambuStudio/wiki/Linux-Compile-Guide)

emelbardis, 2023-03-07:
> There are instructions on the official Bambu Studio repository to build [Linux-Compile-Guide](https://github.com/bambulab/BambuStudio/wiki/Linux-Compile-Guide)

Are you still working on rebuilding the appimage or should we follow the compile instructions?

fang64, 2023-03-08:
> > There are instructions on the official Bambu Studio repository to build [Linux-Compile-Guide](https://github.com/bambulab/BambuStudio/wiki/Linux-Compile-Guide)
> 
> Are you still working on rebuilding the appimage or should we follow the compile instructions?

I am still working on this, I have a day job though that interferes with my ability to just do this, I tend to chase this on the weekend the linux compile guide is fine though.

garethmcc, 2023-03-15:
Maybe this will help. The Arch User Repository has a working compilation of BambuStudio. https://aur.archlinux.org/packages/bambustudio

royw, 2023-06-08:
Also I had to add the following packages to build on kubuntu 23.04:

royw, 2023-07-17:
I just found what was blocking me from running OrcaSlicer on Kubuntu 23.04 (and Ubuntu 20.04 via distrobox).  These were the key messages:

> Gtk-Message: 17:23:32.795: Failed to load module "colorreload-gtk-module"
> Gtk-Message: 17:23:32.795: Failed to load module "window-decorations-gtk-module"

And the cause was the Gtk settings.ini files in /etc/gtk*/ or ~/.config/gtk*/.  I just moved those directories and wow!  OrcaSlicer started!  Ref this post: [Where are "colorreload-gtk-module" and "window-decorations-gtk-module"?](https://www.reddit.com/r/Gentoo/comments/ubenem/where_are_colorreloadgtkmodule_and/)

HTH!

fang64, 2023-07-19:
@royw interesting, that's a little unexpected.

royw, 2023-07-19:
Sorry, I spoke too soon.  After using OrcaSlicer to slice and print a couple of models, I shut my system down.  Well today I'm back to the same failure to launch:

> Fatal glibc error: malloc.c:2593 (sysmalloc): assertion failed: (old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)
Aborted (core dumped)

The ~/.config/gtk-* directories were back.  I deleted them but still no go...

This rabbit hole is deeper yet...

fang64, 2023-07-19:
@royw my current thought is the libraries on the system are different enough to trigger a problem. I have been unable to chase exactly what dependency is causing the hard crashes. One thing that I noticed, is the hard crashes only start after you login which still points me at webkit because that's the browser it uses. But it's really weird it'll run and let you login until that happens.

KenricYoung, 2023-09-15:
Helped tracked resultant issue of this for a user on the fedora discord for the latest 1.7.0 appimage.
Seems to be a decently active linux bug as many similar issues have been created. See #185 #425 #1841 #572

For Workstation users and the Plasma spin, as @magnum61 mentioned, fedora 38 users must install the webkit2gtk4.0 to launch the appimage. 

I used toolbx using the fedora-toolbox:38 container to run the appimage. The container image needed the following packages and their 99 dependencies to launch the appimage.
1. fuse-libs --> libfuse.so.2
2. fontconfig --> libfontconfig.so.1
3. libglvnd-glx -->libGL.so.1
4. libglvnd-egl --> libEGL.so.1
5. gtk3 --> libgtk-3.so.0
6. gstreamer1-plugins-base --> libgstvideo-1.0.so.0
7. webkit2gtk4.0 --> libwebkit2gtk-4.0.so.37

royw, 2023-09-15:
Whoooooop!
I read @KenricYoung 's post so did a quick check that I had his list of prerequisites:

> ➤ whereis libfuse.so.2 libfontconfig.so.1 libGL.so.1 libEGL.so.1 libgtk-3.so.0 libgstvideo-1.0.so.0 libwebkit2gtk-4.0.so.37
> libfuse.so.2: /usr/lib/x86_64-linux-gnu/libfuse.so.2
> libfontconfig.so.1: /usr/lib/x86_64-linux-gnu/libfontconfig.so.1
> libGL.so.1: /usr/lib/x86_64-linux-gnu/libGL.so.1
> libEGL.so.1: /usr/lib/x86_64-linux-gnu/libEGL.so.1
> libgtk-3.so.0: /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
> libgstvideo-1.0.so.0: /usr/lib/x86_64-linux-gnu/libgstvideo-1.0.so.0
> libwebkit2gtk-4.0.so.37: /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37

Tried my mid-August weekly build same crash.  Pulled down the new 1.7.0 release.  It launched!  Cool.  Do the login.  After selecting continue on the "do you want to use Bambu Handy" question, KDE crashed.  Rebooted, OrcaSlicer launched fine, connected to X1C.  Cool!.  Exited and restarted OrcaSlicer.  Still working!  Oh yah!  So reboot, launch OrcaSlicer and all is looking good!  

THANK YOU!!!

Jo-Con-El, 2023-12-05:
@fang64: Did you merge any work with main or this was fixed by someone else?  Is there anything else to fix here?

fang64, 2023-12-06:
> @fang64: Did you merge any work with main or this was fixed by someone else? Is there anything else to fix here?

I got it working originally, what I intended to do with both Bambu Studio and Orca Slicer was fix the appimage to actually be portable, but it may not be needed BS opted to build on Fedora latest and Ubuntu 20.04 to act as a in-between solution. What needs to happen is libwebkit2gtk needs to be shipped in the appimage and I was planning on eventually looking at subsurface ( linus torvalds diving software ) and how they are using linuxdeployqt, because it is better at capturing the libraries that are needed. Although my last efforts chasing this resulted in Segfaults so I didn't see a reason to merge anything yet. @Jo-Con-El

orangemn6, 2023-12-12:
Is this still being worked on? I'm having issues with needing to install the dependencies on Arch.

fang64, 2023-12-12:
@orangemn6 you should be able to use the Fedora AppImage just install webkit2gtk

orangemn6, 2023-12-13:
@fang64 I have done that, it still crashes on launch with many gtk errors

fang64, 2023-12-13:
@orangemn6 can you provide the output of Bambu Studio (the fedora appimage) from console and possibly provide a package list via pacman -Qe

I just spun up a new VM with Arch install with Gnome installed and it ran after installing webkit2gtk. I'm half wondering if it's a missing library.

kinghat, 2023-12-13:
> Helped tracked resultant issue of this for a user on the fedora discord for the latest 1.7.0 appimage. Seems to be a decently active linux bug as many similar issues have been created. See #185 #425 [...]

thanks for your digging! should these be included into the appimage for it to work without the user having to install anything extra?

fang64, 2023-12-13:
> thanks for your digging! should these be included into the appimage for it to work without the user having to install anything extra?

I've tried that approach, you will get an immediate segfault, if you run ldd on those binaries you'll see we pretty much just need to pack up a ton of Ubuntu 18.04 libraries at that point. I'm not defending the build method, but I also don't deny this is a problem...

ChaosBlades, 2024-01-29:
Instead of putting work into AppImage wouldn't it be better to just move to Flatpak?
https://github.com/SoftFever/OrcaSlicer/issues/1670

BambuStudio, PrusaSlicer, and Cura have all moved to Flatpak.
https://flathub.org/apps/com.bambulab.BambuStudio
https://flathub.org/apps/com.prusa3d.PrusaSlicer
https://flathub.org/apps/com.ultimaker.cura

fang64, 2024-01-29:
@ChaosBlades that's a reasonable assessment, I didn't know it was ported to flatpak. AppImage relies on Fuse support being available as well so I agree here. I'll close this out.

closed 2024-01-29 by fang64