OrcaSlicer/OrcaSlicer #233: Spiral Z-Hops and "Avoid Crossing Wall" can combine to make badly ordered hops, causing wall collisions

psiberfunk, 2023-01-25:
**Describe the bug**
In some situations, Using spiral Z-Hops and "Avoid Crossing Wall" (ON) can combine to make badly ordered hops, causing wall collisions.  In these situations the spiral retract happens AFTER the travel move instead of before it!  This negates the whole purpose of the Z-hop and causes significant surface artifacts on the print.

If "Avoid crossing wall" is turned OFF, the planning bug disappears. Alternatively, if normal z-hops are used instead of spiral (but "Avoid crossing wall" is left on) the bug also disappears.  It is only when BOTH are on that the bug seems to happen in this example. 

**3mf File for This Bug**
Pre-sliced and unsliced files are provided here:
[3MF_Files_AvoidCrossingWallsSpiralZhopGlitch.zip](https://github.com/SoftFever/BambuStudio-SoftFever/files/10501984/3MF_Files_AvoidCrossingWallsSpiralZhopGlitch.zip)

**To Reproduce**
Scanning through the GCODE (see movie attached here) shows the problem is that the order of operations is wrong:
1)Wipe
2)Travel move to desired location
3)Spiral hop (retract)
4)Unretract    [in the same spot as the retraction just happened!]

This is incorrect,   the right ordering of operations  is   

1)Wipe
2)Spiral hop (retract)
3)Travel move to desired location
4)Unretract    

**Movie showing bad order**
https://user-images.githubusercontent.com/19777571/214640259-a998e8e1-3c59-4d33-aa83-fa079aaee9a4.mp4

**Desktop (please complete the following information):**
 - MacOS
 - SoftFever 1.4.2

SoftFever, 2023-02-02:
Yep, it's indeed a bug.

SoftFever, 2023-02-03:
This should fix the issue: https://github.com/SoftFever/BambuStudio-SoftFever/pull/279
You can download the build here: https://github.com/SoftFever/BambuStudio-SoftFever/actions

psiberfunk, 2023-02-24:
This is verified as fixed

closed 2023-02-24 by psiberfunk