OrcaSlicer/OrcaSlicer #235: Locale related crash on startup (Arch build from source)

pfrench42, 2023-01-25:
**Describe the bug**
A clear and concise description of what the bug is.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Compile the program from source.
run the program it crashes while loading the locale, before the UI comes up

**Expected behavior**
Main application window should come up

**Screenshots**
This is the backtrace of the segfault
Notice in frame #3, linux_get_existing_locale_laguage is being passed NULL for system_language, and crashes when doing member accesses of system_language

This is called from GUI_App::load_language() where m_language_info_system is NULL
   language_info = linux_get_existing_locale_language(language_info, m_language_info_system);
(gdb) bt
```
#0  std::__cxx11::basic_string<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> >::find(wchar_t, unsigned long) const (this=0x8, __c=95 L'_', __pos=0)
    at /usr/src/debug/gcc/gcc-build/x86_64-pc-linux-gnu/libstdc++-v3/include/bits/basic_string.tcc:687
#1  0x000055555984cd7f in wxString::Find(wxUniChar, bool) const ()
#2  0x000055555984cf9e in wxString::AfterFirst(wxUniChar) const ()
#3  0x0000555556d0a6b5 in Slic3r::GUI::linux_get_existing_locale_language(wxLanguageInfo const*, wxLanguageInfo const*) (language=0x55555c515800, system_language=0x0)
    at /home/bak/pfrench/git/3dprint/BambuStudio-SoftFever/src/slic3r/GUI/GUI_App.cpp:4282
#4  0x0000555556d0caa2 in Slic3r::GUI::GUI_App::load_language(wxString, bool) (this=0x55555c299f30, language=..., initial=true)
    at /home/bak/pfrench/git/3dprint/BambuStudio-SoftFever/src/slic3r/GUI/GUI_App.cpp:4499
#5  0x0000555556cf8cba in Slic3r::GUI::GUI_App::on_init_inner() (this=0x55555c299f30) at /home/bak/pfrench/git/3dprint/BambuStudio-SoftFever/src/slic3r/GUI/GUI_App.cpp:2116
#6  0x0000555556cf6faf in Slic3r::GUI::GUI_App::OnInit() (this=0x55555c299f30) at /home/bak/pfrench/git/3dprint/BambuStudio-SoftFever/src/slic3r/GUI/GUI_App.cpp:2009
#7  0x000055555981ab53 in wxEntry(int&, wchar_t**) ()
#8  0x0000555556ce671c in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) (params=...) at /home/bak/pfrench/git/3dprint/BambuStudio-SoftFever/src/slic3r/GUI/GUI_Init.cpp:64
#9  0x0000555555cb307e in Slic3r::CLI::run(int, char**) (this=0x7fffffffd620, argc=1, argv=0x7fffffffd858) at /home/bak/pfrench/git/3dprint/BambuStudio-SoftFever/src/BambuStudio.cpp:452
#10 0x0000555555cc980e in main(int, char**) (argc=1, argv=0x7fffffffd858) at /home/bak/pfrench/git/3dprint/BambuStudio-SoftFever/src/BambuStudio.cpp:2623
```

If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Linux (Manjaro)
 - Version 5.15.89-1-MANJARO #1 SMP PREEMPT

pfrench42, 2023-01-25:
patch which fixed this for me:
```
diff --git a/src/slic3r/GUI/GUI_App.cpp b/src/slic3r/GUI/GUI_App.cpp
index 3948589b..d830229a 100644
--- a/src/slic3r/GUI/GUI_App.cpp
+++ b/src/slic3r/GUI/GUI_App.cpp
@@ -4496,6 +4496,9 @@ bool GUI_App::load_language(wxString language, bool initial)
     // instead of just reporting that it is impossible to switch.
     if (! wxLocale::IsAvailable(language_info->Language)) {
         std::string original_lang = into_u8(language_info->CanonicalName);
+        if ( nullptr == m_language_info_system ) {
+            m_language_info_system = wxLocale::GetLanguageInfo(wxLANGUAGE_DEFAULT);
+        }
         language_info = linux_get_existing_locale_language(language_info, m_language_info_system);
         BOOST_LOG_TRIVIAL(trace) << boost::format("Can't switch language to %1% (missing locales). Using %2% instead.")
                                     % original_lang % language_info->CanonicalName.ToUTF8().data();
```

SoftFever, 2023-01-26:
Great. Thanks 
Do you want to create a PR for the fix?

pfrench42, 2023-01-26:
I was not planning to.  I've not even created a fork of my own. 
Great SW, BTW, thanks for doing this.
If you want me to fork and submit I'll do it but this was a pretty small change.

woolmonkey, 2023-03-25:
Any help on compiling it?  Keep trying and it stops with errors all the time.

pwhelan, 2023-08-08:
Does this still need to be merged?

pfrench42, 2023-08-10:
I'd love to see it merged.  As-is the AppImage crashes every time on my
Arch distro

pwhelan, 2023-08-10:
> I'd love to see it merged. As-is the AppImage crashes every time on my Arch distro

I was able to fix it by setting my language manually .config/OrcaSlicer/OrcaSlicer.conf. I cannot recall or even find where I set it. You could try erasing that configuration file and seeing if then it can execute correctly.

SoftFever, 2023-08-11:
A patch has been submitted based on this.
Let me know if it fixes the issue.
The build will be ready in about half an hour: https://github.com/SoftFever/OrcaSlicer/actions/runs/5827305845

> patch which fixed this for me:
>
> ```
> diff --git a/src/slic3r/GUI/GUI_App.cpp b/src/slic3r/GUI/GUI_App.cpp
> index 3948589b..d830229a 100644
> --- a/src/slic3r/GUI/GUI_App.cpp
> +++ [...]

pwhelan, 2023-08-16:
It works for me.

pfrench42, 2023-08-16:
What's interesting is the AppImage downloaded still crashes at startup, but
I'm not sure if it's the same thing.

I've forked the repo and if I
* build from source (in DEBUG) mode
* running orca-slicer from <repo>/build/package/ doesn't crash at start,
but does crash during slicing (Press "Preview")
* although I do get "Error running JavaScript" Unsupported result type",
but it's non-fatal

I'll bring it up under gdb and see what I get

pfrench42, 2023-08-16:
backtrace:
orca-slicer:
/home/pfrench/proj/github/OrcaSlicer/src/libslic3r/PerimeterGenerator.cpp:406:
Slic3r::clip_extrusion(const ClipperLib_Z::Path&, const
ClipperLib_Z::Paths&, ClipperLib_Z::ClipType)::<lambda(const
ClipperLib_Z::IntPoint&, const ClipperLib_Z::IntPoint&, const
ClipperLib_Z::IntPoint&, const ClipperLib_Z::IntPoint&,
ClipperLib_Z::IntPoint&)>: Assertion `start.z() > 0 && end.z() > 0' failed.

Thread 802 "slic3r_tbb_18" received signal SIGABRT, Aborted.
[Switching to Thread 0x7ffdecff36c0 (LWP 1216404)]
0x00007ffff008e83c in ?? () from /usr/lib/libc.so.6
(gdb) bt
#0  0x00007ffff008e83c in  () at /usr/lib/libc.so.6
#1  0x00007ffff003e668 in raise () at /usr/lib/libc.so.6
#2  0x00007ffff00264b8 in abort () at /usr/lib/libc.so.6
#3  0x00007ffff00263dc in  () at /usr/lib/libc.so.6
#4  0x00007ffff0036d26 in  () at /usr/lib/libc.so.6
#5  0x0000555556a0b64f in operator()(ClipperLib_Z::IntPoint const&,
ClipperLib_Z::IntPoint const&, ClipperLib_Z::IntPoint const&,
ClipperLib_Z::IntPoint const&, ClipperLib_Z::IntPoint&) const
(__closure=0x7ffdecfe6b10, e1bot=..., e1top=..., e2bot=..., e2top=...,
pt=...)
    at
/home/pfrench/proj/github/OrcaSlicer/src/libslic3r/PerimeterGenerator.cpp:406
#6  0x0000555556a16a21 in std::__invoke_impl<void,
Slic3r::clip_extrusion(const ClipperLib_Z::Path&, const
ClipperLib_Z::Paths&, ClipperLib_Z::ClipType)::<lambda(const
ClipperLib_Z::IntPoint&, const ClipperLib_Z::IntPoint&, const
ClipperLib_Z::IntPoint&, const ClipperLib_Z::IntPoint&,
ClipperLib_Z::IntPoint&)>&, const Eigen::Matrix<int, 3, 1, 2, 3, 1>&, const
Eigen::Matrix<int, 3, 1, 2, 3, 1>&, const Eigen::Matrix<int, 3, 1, 2, 3,
1>&, const Eigen::Matrix<int, 3, 1, 2, 3, 1>&, Eigen::Matrix<int, 3, 1, 2,
3, 1>&>(std::__invoke_other, struct {...} &) (__f=...) at
././/include/c++/13.2.1/bits/invoke.h:61
#7  0x0000555556a16201 in std::__invoke_r<void,
Slic3r::clip_extrusion(const ClipperLib_Z::Path&, const
ClipperLib_Z::Paths&, ClipperLib_Z::ClipType)::<lambda(const
ClipperLib_Z::IntPoint&, const ClipperLib_Z::IntPoint&, const
ClipperLib_Z::IntPoint&, const ClipperLib_Z::IntPoint&,
ClipperLib_Z::IntPoint&)>&, const Eigen::Matrix<int, 3, 1, 2, 3, 1>&, const
Eigen::Matrix<int, 3, 1, 2, 3, 1>&, const Eigen::Matrix<int, 3, 1, 2, 3,
1>&, const Eigen::Matrix<int, 3, 1, 2, 3, 1>&, Eigen::Matrix<int, 3, 1, 2,
3, 1>&>(struct {...} &) (__fn=...) at
././/include/c++/13.2.1/bits/invoke.h:111
#8  0x0000555556a15acf in std::_Function_handler<void(const
Eigen::Matrix<int, 3, 1, 2, 3, 1>&, const Eigen::Matrix<int, 3, 1, 2, 3,
1>&, const Eigen::Matrix<int, 3, 1, 2, 3, 1>&, const Eigen::Matrix<int, 3,
1, 2, 3, 1>&, Eigen::Matrix<int, 3, 1, 2, 3, 1>&),
Slic3r::clip_extrusion(const ClipperLib_Z::Path&, const
ClipperLib_Z::Paths&, ClipperLib_Z::ClipType)::<lambda(const
ClipperLib_Z::IntPoint&, const ClipperLib_Z::IntPoint&, const
ClipperLib_Z::IntPoint&, const ClipperLib_Z::IntPoint&,
ClipperLib_Z::IntPoint&)> >::_M_invoke(const std::_Any_data &, const
Eigen::Matrix<int, 3, 1, 2, 3, 1> &, const Eigen::Matrix<int, 3, 1, 2, 3,
1> &, const Eigen::Matrix<int, 3, 1, 2, 3, 1> &, const Eigen::Matrix<int,
3, 1, 2, 3, 1> &, Eigen::Matrix<int, 3, 1, 2, 3, 1> &) (__functor=...,
__args#0=..., __args#1=..., __args#2=..., __args#3=..., __args#4=...)
    at ././/include/c++/13.2.1/bits/std_function.h:290
#9  0x0000555557e99566 in std::function<void (Eigen::Matrix<int, 3, 1, 2,
3, 1> const&, Eigen::Matrix<int, 3, 1, 2, 3, 1> const&, Eigen::Matrix<int,
3, 1, 2, 3, 1> const&, Eigen::Matrix<int, 3, 1, 2, 3, 1> const&,
Eigen::Matrix<int, 3, 1, 2, 3, 1>&)>::operator()(Eigen::Matrix<int, 3, 1,
2, 3, 1> const&, Eigen::Matrix<int, 3, 1, 2, 3, 1> const&,
Eigen::Matrix<int, 3, 1, 2, 3, 1> const&, Eigen::Matrix<int, 3, 1, 2, 3, 1>
const&, Eigen::Matrix<int, 3, 1, 2, 3, 1>&) const
    (this=0x7ffdecfe6b10, __args#0=..., __args#1=..., __args#2=...,
__args#3=..., __args#4=...) at
././/include/c++/13.2.1/bits/std_function.h:591
#10 0x0000555557e899aa in ClipperLib_Z::Clipper::SetZ(Eigen::Matrix<int, 3,
[... 96 lines cut ...]
tbb::detail::d1::execution_data &) (this=0x7fff5801c3e8, start=warning:
RTTI symbol not found for class
'tbb::detail::d1::start_for<tbb::detail::d1::blocked_range<unsigned long>,
Slic3r::PrintObject::make_perimeters()::{lambda(tbb::detail::d1::blocked_range<unsigned
long> const&)#2}, tbb::detail::d1::auto_partitioner const>'
..., range=..., ed=...)
    at
/home/pfrench/proj/github/OrcaSlicer/deps/build/destdir././/local/include/oneapi/tbb/partitioner.h:288
#24 0x000055555647044c in
tbb::detail::d1::start_for<tbb::detail::d1::blocked_range<long unsigned
int>, Slic3r::PrintObject::make_perimeters()::<lambda(const
tbb::detail::d1::blocked_range<long unsigned int>&)>, const
tbb::detail::d1::auto_partitioner>::execute(tbb::detail::d1::execution_data
&) (this=0x7fff5801c380, ed=...)
    at
/home/pfrench/proj/github/OrcaSlicer/deps/build/destdir././/local/include/oneapi/tbb/parallel_for.h:172
#25 0x00005555599f4607 in tbb::detail::r1::market::process(rml::job&)
(this=0x1286ec, j=...)
    at
/home/pfrench/proj/github/OrcaSlicer/deps/build/dep_TBB-prefix/src/dep_TBB/src/tbb/task_dispatcher.h:322
#26 0x00005555599efa40 in thread_routine(void*) (arg=0x55555f65c380)
    at
/home/pfrench/proj/github/OrcaSlicer/deps/build/dep_TBB-prefix/src/dep_TBB/src/tbb/private_server.cpp:267
#27 0x00007ffff008c9eb in  () at /usr/lib/libc.so.6
#28 0x00007ffff011123c in  () at /usr/lib/libc.so.6

closed 2023-11-25 by github-actions[bot]