OrcaSlicer/OrcaSlicer #240: Media playback error: No URL-handler implemented for "bambu" in Ubuntu 22.10

RobRobM, 2023-01-27:
**Describe the bug**
When pressing the play button on the Device Tab the message Media playback error: No URL-handler implemented for "bambu" displays and the camera feed does not work

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Device tab
2. Click on the play button
4. See error

**Expected behavior**
The camera feed should start.

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Version 22.10

SoftFever, 2023-01-27:
The latest 1.4.3 version should have the video now.

RobRobM, 2023-01-29:
720p works! With 1080p resolution I only get a black screen. And the default settings is 1080p. So to get the camera working I had to switch to 720p and restart the stream to get it working.

Great work by the way. This project is the reason I bought the X1C!!! 

Thanks for all your hard work!!!!!

closed 2023-02-01 by RobRobM