OrcaSlicer/OrcaSlicer #241: Softfever 1.4.3 - PA Lines are missing since new Version

tommy536, 2023-01-27:
I didnt know its a bug or a feature.

When i began to setup a PA Test. Slice it, there comes out a very short print within 3 Seconds and the micro line. There are no PA lines showing. I came from 1.4.2. I tested it backwards with 1.4.2 and there it works normaly.

SoftFever, 2023-01-27:
Hi,
What printer did you use? Did you change any settings in the PA dialog? 

if possible, share the gcode file will be very helpful

tommy536, 2023-01-27:
Hi,
attached the Gcode File. I am using a Voron 2.4.

[image]

[image]

[pressure_advance_test_PA-CF_3s.zip](https://github.com/SoftFever/BambuStudio-SoftFever/files/10521430/pressure_advance_test_PA-CF_3s.zip)

hardtech87, 2023-01-27:
i have the same problem. X1c Combo.
version 1.4.2 works fine

hliebscher, 2023-01-28:
the arm64 version at Mac 13.1 Ventura is working i think the x86_64 Mac version needs rebuild with clear build directory

hliebscher, 2023-01-28:
@SoftFever PA Test works only in ENGLISH LANGUAGE ;-) in German no output !!!!!

SoftFever, 2023-01-28:
Ouch.
Looks like only English and Chinese works.
I will note it in the release notes for now while I'm investigating the issue.

> @SoftFever PA Test works only in ENGLISH LANGUAGE ;-) in German no output !!!!!

dimitrigerber, 2023-01-28:
[image]

Hello, the window ist pretty small so I cant see all values.

SoftFever, 2023-01-28:
> 

Which OS and resolution do you use?

dimitrigerber, 2023-01-28:
[image]

dimitrigerber, 2023-01-28:
Windows resolution 3000x2000 and 200% scaling

SoftFever, 2023-01-28:
> 

Ah, okay, DPI stuff.
Will fix it in the coming hotfix release

SoftFever, 2023-01-28:
Thanks for reporting the bug.
The problem has been resolved in the latest hotfix release.

hliebscher, 2023-01-28:
@tommy536  if is fine for you can you close the bug ?

closed 2023-05-06 by SoftFever