OrcaSlicer/OrcaSlicer #243: SoftFever 1.4.3- MAC OS launch with errors

christianjou, 2023-01-27:
**Describe the bug**
When i launch SoftFever 1.4.3 there is some errors and it halts

**To Reproduce**
Steps to reproduce the behavior:
1. JUST LAUNCH

**Screenshots**
[image]

Full text errors : 

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               BambuStudio [10025]
Path:                  /Users/USER/Downloads/BambuStudio-SoftFever 3.app/Contents/MacOS/BambuStudio
Identifier:            com.bambulab.bambu-studio
Version:               01.04.02.13 (???)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-01-27 17:25:44.1916 +0100
OS Version:            macOS 12.6.3 (21G419)
Report Version:        12
Bridge OS Version:     3.0 (14Y910)
Anonymous UUID:        83E42FDE-A5FA-6BF6-F29A-D22600391A91

Sleep/Wake UUID:       41E6A0B7-07E2-4E19-9694-99C057F3BB73

Time Awake Since Boot: 25000 seconds
Time Since Wake:       1033 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: '/usr/local/opt/zstd/lib/libzstd.1.dylib'
Referenced from: '/Users/USER/Downloads/BambuStudio-SoftFever 3.app/Contents/MacOS/BambuStudio'
Reason: tried: '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '' (no such file), '' (no such file)
(terminated at launch; ignore backtrace)

Thread 0 Crashed:
0   dyld                          	       0x112c6a98e __abort_with_payload + 10
1   dyld                          	       0x112c80a72 abort_with_payload_wrapper_internal + 80
2   dyld                          	       0x112c80aa4 abort_with_payload + 9
3   dyld                          	       0x112c2e699 dyld4::halt(char const*) + 615
4   dyld                          	       0x112c2a5b5 dyld4::prepare(dyld4::APIs&, dyld3::MachOAnalyzer const*) + 3995
5   dyld                          	       0x112c294e4 start + 388

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000002000209  rbx: 0x0000000000000000  rcx: 0x00007ff7b707f538  rdx: 0x00007ff7b707f9a0
  rdi: 0x0000000000000006  rsi: 0x0000000000000001  rbp: 0x00007ff7b707f580  rsp: 0x00007ff7b707f538
   r8: 0x00007ff7b707f5a0   r9: 0x0000000000000000  r10: 0x0000000000000093  r11: 0x0000000000000246
  r12: 0x0000000000000093  r13: 0x00007ff7b707f9a0  r14: 0x0000000000000001  r15: 0x0000000000000006
  rip: 0x0000000112c6a98e  rfl: 0x0000000000000246  cr2: 0x0000000112c30da4
  
Logical CPU:     0
Error Code:      0x02000209 
Trap Number:     133

Binary Images:
       0x112c24000 -        0x112c8ffff dyld (*) <006a3e6f-3cd3-34d9-b0f2-ed6bd67a95a6> /usr/lib/dyld
[... 114 lines cut ...]
    }
  ],
  "experiments" : [

  ]
}
}

Model: MacBookPro13,2, BootROM 499.40.2.0.0, 2 processors, Dual-Core Intel Core i5, 2,9 GHz, 8 GB, SMC 2.37f25
Graphics: Intel Iris Graphics 550, Intel Iris Graphics 550, Built-In
Display: Color LCD, 2560 x 1600 Retina, Main, MirrorOff, Online
Memory Module: BANK 0/DIMM0, 4 GB, LPDDR3, 2133 MHz, 0x80CE, 0x4B3445364533303445422D45474347202020
Memory Module: BANK 1/DIMM0, 4 GB, LPDDR3, 2133 MHz, 0x80CE, 0x4B3445364533303445422D45474347202020
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x157), Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1710.4)
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB30Bus
USB Device: Apple T1 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.5
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.5

**Desktop (please complete the following information):**
 - OS:  MONTEREY
 -  - Version 12.6.3

hliebscher, 2023-01-28:
@SoftFever 
Mac M1 (arm64) is working
the x86_64 version from github has the bug
my own self build version 1.4.3 is working.

Manfredo1952, 2023-01-28:
@SoftFever  @hliebscher 
My Intel-Version is starting flawless. No  problems.

Manfredo1952, 2023-01-28:
Error report says "
[image]

SoftFever, 2023-01-28:
An error with the host runner's dependency was identified in the CI/CD build.
A new build has been uploaded to correct the issue.
Please download the updated build from the release page.

christianjou, 2023-01-28:
All is perfect now. Thanks.

hliebscher, 2023-01-28:
@christianjou pls close the issue if is fine

christianjou, 2023-01-28:
done

closed 2023-01-28 by christianjou