OrcaSlicer/OrcaSlicer #244: 1.4.3 Perimeter printing order still wrong is some case

ShryuKt, 2023-01-27:
I had hope when I saw

> Fixes:

> Wrong perimeter generation for overhangs in some cases https://github.com/bambulab

My problem is still present in 1.4.3
I have already referenced it on the bambu studio repo [BambuStudio#1055](https://github.com/bambulab/BambuStudio/issues/1055)
But maybe you can help

In some case the outer perimeters are still printed first
[image]
On this part, these two holes are overhanging, which makes the printing fail because the outer perimeters are printed on a void without support.
All other holes are printed in the correct order.

hliebscher, 2023-01-27:
its the same error on PrusaSlicer 2.5.0  and SuperSlicer
[image]

ShryuKt, 2023-01-28:
I use superslicer 2.4.58.5 (so before arachne) for everydays prints and the order is correct
This is my 3mf with SS, same piece and almost same settings, as far as possible, as with BS
[Tap_Front_r2.SS.3mf.txt](https://github.com/SoftFever/BambuStudio-SoftFever/files/10526639/Tap_Front_r2.SS.3mf.txt)

[image]

Maybe this is something that appended with the implementation of arachne wall generator.

SoftFever, 2023-01-28:
It's an edge case here.
The thing is: if you observe those wall loops carefully, the hole has only one wall.
Other walls/perimeters belong to the contour.

In the case of Arachne, the hole has 0 walls on its own; hence it solved the issue accidently.

We should be able to special handling such cases. 
Stay tuned.

ShryuKt, 2023-01-28:
> We should be able to special handling such cases. Stay tuned.

That would be great,
Thank you for taking this into consideration

julie777, 2023-05-31:
I just added quite a bit of information to #884 which also applies here.

closed 2023-11-25 by github-actions[bot]