OrcaSlicer/OrcaSlicer #246: Hi

Theos88, 2023-01-27:
**Describe the bug**
A clear and concise description of what the bug is.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

Theos88, 2023-01-27:
Hi,

the PA Calibration Tower Method does not work in my case. It seems to print with one fix value the whole print through. In V1.4.2 it works just fine as intended. Do you need further input? I didn´t test the Line Method.

hliebscher, 2023-01-27:
for Mac it works 
;AFTER_LAYER_CHANGE
;45
**M572 D0 S0.09; Override pressure advance value**
; printing object tower_with_seam.stl id:0 copy 0
M204 P800 ; adjust acceleration

Theos88, 2023-01-28:
I´ll post the gcode later. Only difference I noticed so far is, that my gcode uses M900 Linear Advance instead of M572 Pressure Advance.

hliebscher, 2023-01-28:
@Theos88   M572 is for my Printer MARLIN needs M900 with growing up parameters each line

SoftFever, 2023-01-28:
@Theos88 
Confirmed, it was a bug.
Switch to line mode then switch back to tower mode in the PA dialog will fix it.
I will release a hotfix soon

Theos88, 2023-01-28:
Ok, thank you!

closed 2023-01-28 by Theos88

SoftFever, 2023-01-28:
Thanks for reporting the bug.
The problem has been resolved in the latest hotfix release.

ryguyy46, 2023-06-12:
I’m having the same issue. PA Tower seems to print with the same value, I tired using line and and all of them looked exactly the same. Even tried switching back and forth from line to tower. Same results.

gamelover42, 2023-07-28:
I just sliced and printed the PA tower using v1.6.3 and the corners are identical all the way up.