OrcaSlicer/OrcaSlicer #247: LED Lights

Highlander-bdj, 2023-01-27:
The P1P start GCODE contains code to turn on the LEDs: M960 S5 P1 ; turn on logo lamp

Would you add code to the end GCODE to turn them off?

hliebscher, 2023-01-27:
M960 S5 P0/1 | Bambu Lab toolhead logo light off/on

@Highlander-bdj  pls add yourself the code M960 S5 P0 to your END CODE

Highlander-bdj, 2023-01-27:
Thank you for that. Would you consider incorporating it into the default. end code since the on is already there?  I know it’s minor but doing it myself makes me have to create a custom profile.

hliebscher, 2023-01-28:
@Highlander-bdj pls send me the tested END GCODE i make a PR for the Profile

hliebscher, 2023-02-11:
@Highlander-bdj can you close the bug if you fix this?

Highlander-bdj, 2023-02-11:
I thought I did?

closed 2023-02-11 by Highlander-bdj