OrcaSlicer/OrcaSlicer #249: *Feature Request*Please allow inches/mm toggle like PrusaSlicer

Photogad, 2023-01-28:
Posted in official BambuStudio git but no traction.
https://github.com/bambulab/BambuStudio/issues/946

> ok I realize you can switch MM or Inches in the BambuSlicer main settings but it's much more convenient the way it is in PrusaSlicer with the checkbox under the input boxes. Sometimes I just want to quickly size something to 8 inches long on Y axis (the women love this size) and not have to to get rid of MM for everything else.
This was a feature from PrusaSlicer you should not have removed, it's very convenient

hliebscher, 2023-01-28:
pls make the request also by BambuStudio

SoftFever, 2023-02-03:
> 

I had a hard time finding a good place to put that checkbox...
Maybe a keyboard shortcut instead?

Photogad, 2023-05-14:
> > 
> 
> I had a hard time finding a good place to put that checkbox... Maybe a keyboard shortcut instead?

@SoftFever 
I think if we could do it like how they have it in the PrusaSlicer or possibly in the SuperSlicer it would work well.

closed 2023-11-25 by github-actions[bot]