OrcaSlicer/OrcaSlicer #256: Brim function is broken

YyAoMmIi, 2023-01-29:
[image]

This is certainly more than 3mm. i tested with outer only, and still get the issue. inner only did not brim anything at all.
Also, this brim is weird since a brim should not overlap the 1st layer, but it does so here.

hliebscher, 2023-01-29:
for me it works at MAC M1
[image]

hliebscher, 2023-01-29:
@YyAoMmIi  can you send the model or a link?

YyAoMmIi, 2023-01-29:
https://www.myminifactory.com/object/3d-print-flexi-print-in-place-imperial-dragon-218337

hliebscher, 2023-01-29:
pls send your 3mf file with settings with 1.4.4 works

hliebscher, 2023-02-11:
@YyAoMmIi can you close the bug if this works?

YyAoMmIi, 2023-02-11:
I had had sent the file to softfever on discord since I didn't have a way
to contact you privately .
I think the issue is more when there multiple plate. when i was printing
from plate 1-11 it was fine, but plate 12 and it broke.

closed 2023-02-11 by YyAoMmIi