OrcaSlicer/OrcaSlicer #257: Slim tree supports printing too thin/low flow?

Arakon, 2023-01-29:
**Describe the bug**
When printing with the new Slim Tree supports, the support has gaps and seems to be underextruded. The model outside walls printed at the same speeds or even faster do not exhibit this issue.

**3mf File for This Bug**
[astronaut_hollow_moon.zip](https://github.com/SoftFever/BambuStudio-SoftFever/files/10529330/astronaut_hollow_moon.zip)

**To Reproduce**
Steps to reproduce the behavior:
Print the attached file.

**Expected behavior**
Similar to Cura tree supports, the supports should have a solid outer shell.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - Windows 10

hliebscher, 2023-01-29:
on my ratrig with 0.4 Nozzle it woks

davidja92, 2023-01-29:
confirmed works on my X1C with .4 nozzle. double check all your settings, what filament are you using? possibly under extrusion due to small layer time, too high heat for that layer?

hliebscher, 2023-01-29:
it looks like PLA but your TEMP is 250![image]

davidja92, 2023-01-29:
yea 250 for pla is really high the X1 sets it to 240 for the extras speed lower your temp and try again. high temp with poor cooling with that small layer time will cause your issue

Arakon, 2023-01-29:
The running print was at 220°, not 250°. Probably had accidentally switched to my ASA profile before saving the 3MF. The model itself looked absolutely flawless, ONLY the supports were falling apart.
I reprinted it with Cura's tree supports and they came out perfectly fine too.

zviratko, 2023-01-29:
I just noticed the same today, pretty sure it's filament related (but the model itself printed perfectly). I used the colorFabb Vibers PLA today and had to set supports to 2 walls for them to not lose cohesion. I think supports should have an option to print every single layer, or at least reuse the "Infill Combination" setting and print as normal when not enabled.

SoftFever, 2023-01-30:
> 

I also suspect it could be a bug when using 0.6 mm nozzle.
@zviratko are you using 0.4 or 0.6 nozzle?

zviratko, 2023-01-30:
0.4mm. I also noticed a line on the same print where it underextruded. This is filament related, but accentiated by the dual-layer support perimeter...

I started printing the same model today and there I went for 2 walls and it still has holes. 0.6mm nozzle should work better there...

zviratko, 2023-01-30:
So is it underextruding by a factor of 2, or is it just not compensated? My supports failed again, even with 2 walls :-(

SoftFever, 2023-01-31:
I will push a fix tonight that temporarily disables adaptive layer height for tree support.

hliebscher, 2023-02-01:
for me its working with 0.4 mm nozzle i have no problems with variable layer height
i use tree with hybrid

CraftyAF, 2023-03-04:
Any update on this? It's still very much an issue for me. You can see in the attached screenshots that it doesn't even recognize different layer heights in the adaptive layers of the supports.
[image]
[image]
[image]
[image]

zviratko, 2023-03-05:
This wasn't an issue for me recently, but I'm not sure how many adaptive layer height prints with tree supports I made. Previously, it was broken even with constant height, at least that now works just fine.

CraftyAF, 2023-03-05:
> This wasn't an issue for me recently, but I'm not sure how many adaptive layer height prints with tree supports I made. Previously, it was broken even with constant height, at least that now works just fine.

Are you using slim tree supports? I don't have adaptive layer height on. This only happens with slim tree.

SoftFever, 2023-03-05:
After checking the codes, I can't find what's wrong.
The slim tree has adaptive height always on to save printing time. But that's fine

The attached project will print the supports with 0.6 line width and 0.4mm layer height as the max allowed layer height is set to 0.4 mm. I suspect it's could be and max volumetric speed issue, the hotend. is struggling to push filament at this speed.
Maybe try reducing the max volumetric speed and max layer height?

SoftFever, 2023-03-06:
Anyway, I've pushed a new change that allows turning off adaptive layer height for tree support.
The option is located in `support`->`advance`.

CraftyAF, 2023-03-06:
Awesome!

CraftyAF, 2023-03-07:
> 

Sorry. I missed this reply before. I'm using a reasonable 15mm^3/sec, but I'm printing at .08mm layer height, so that translates to very high speed, which still prints fine. It's definitely a max volumetric speed issue. On the supports, it's not obeying the max volumetric speed limitation with the adaptive layers. You can see in the screenshots that I posted that it doesn't even recognize that it's printing variable layer heights or that it's printing variable flow. It seems to think that it's still printing at .08 layer height and so it's improperly calculating its flow and therefor not reducing speed to obey flow limits.

closed 2023-03-17 by SoftFever

ShantanuNair, 2025-04-05:
@SoftFever I think with layer heights turned on, speedcapping isn't working right for tree supports. Flows are consistently higher than max flow, esp when using high layer heights. If you can point me to where the code for tree supports is, I can start looking at the logic.