OrcaSlicer/OrcaSlicer #259: Pressure advance in gcode notthe same than in setting

DzzD, 2023-01-29:
There is probably something I did not understand but setting pressure advance to 0.036 in filament settings make that gcode is populated with `M900 K0.040`

[image]

**Gcode :**
```
    M900 L1000.0 M1.0
    M900 K0.040 
```
Is that normal ?

SoftFever, 2023-01-29:
This command is certainly not generated by the slicer.
It must be set by your start g-code.
You can search the keyword `0.034` to find the pressure advance instruction generated by the slicer. It comes with comments.

DzzD, 2023-01-29:
I did and only found it in a comment : 
`; pressure_advance = 0.02,0.02,0.034,0.034`

hliebscher, 2023-01-29:
@DzzD you print not with filament 3 or 4 ! pls add PA to filament 1 and 2

DzzD, 2023-01-29:
ha sorry, stupid mistake....

```
M900 K0.015 ; Set linear advance
M900 K0.034 L1000 M10 ; Override pressure advance value
```
What does `L1000 M10` are for ?

I just would like to use that in the original BambuStudio in filament gcode, so I try to understand, I found marlin doc about M900 but they dont talk a lot on L & M values

hliebscher, 2023-01-29:
@DzzD The L and M variables are related to the way the printer does the auto-compensation.
In this case, if you are looking to use a simple value for the compensation, we recommend using M900 K M command
For example: M900 K0.02 L1000 M10 will make sure that the compensation will work in linear mode and the compensation value will not be limited unwillingly, by also applying a 0.02 value for the compensation.

DzzD, 2023-01-30:
Thanks a lot for those explanation I assume I can use on filament setting gcode section by setting `M900 K0.02 L1000 M10` to set linear advance for each filament with only modifying the K factor ?

closed 2023-11-25 by github-actions[bot]