OrcaSlicer/OrcaSlicer #261: Acceleration settings not being applied

DarthMaledous, 2023-01-29:
I ran a couple of test prints with the latest version of Soft Fever and ran into a weird issue. No matter what I do, it won't print above 500mm/s acceleration. I took a look at the gcode in the file it sliced and I can see where it's hard setting the acceleration to 500mm/s. 

[image]
[image]

Snippet of the GCODE that's being produced:

; EXECUTABLE_BLOCK_START
M73 P0 R23
;TYPE:Custom
G90 ; use absolute coordinates
M83 ; extruder relative mode
M140 S60 ; set final bed temp
M104 S150 ; set temporary nozzle temp to prevent oozing during homing
G4 S10 ; allow partial nozzle warmup
G28 ; home all axis
G29
;BED_MESH_PROFILE LOAD=default
G1 Z50 F240
G1 X2 Y10 F3000
M104 S210 ; set final nozzle temp
M190 S60 ; wait for bed temp to stabilize
M109 S210 ; wait for nozzle temp to stabilize
G1 Z0.28 F240
G92 E0
G1 Y140 E10 F1500 ; prime the nozzle
M73 P1 R23
G1 X2.3 F5000
G92 E0
M73 P2 R23
G1 Y10 E10 F1200 ; prime the nozzle
G92 E0
M190 S60 ; set bed temperature and wait for it to be reached
M104 S210 ; set nozzle temperature
G90
G21
M83 ; use relative distances for extrusion
; filament start gcode
M106 S0 ; disable fan
;LAYER_CHANGE
;Z:0.2
;HEIGHT:0.2
G1 E-.85 F3000 ; retract
G1 Z.32 F9000 ; lift Z
SET_VELOCITY_LIMIT ACCEL_TO_DECEL=250 ; adjust max_accel_to_decel to chosen % of new accel value
M204 S500 ; adjust acceleration

You can see the M204 S500 being set to adjust acceleration.

davidja92, 2023-01-29:
@DarthMaledous i went through a sliced gcode im running and yes the first layer sets acceleration to 500. can you please check the start of layer 2 in the gcode to see if acceleration is ramped up? you have first layer accel set to 500

DarthMaledous, 2023-01-29:
Thanks for replying.  Looking the the rest of the sliced gcode,  I see no other references to M204 S500, nor do I see the acceleration being ramped up.  I'm attaching a copy of the gcode.
[square_PLA_23m41s.gcode.txt](https://github.com/SoftFever/BambuStudio-SoftFever/files/10530524/square_PLA_23m41s.gcode.txt)

DarthMaledous, 2023-01-29:
UPDATE: I found the issue.  I had the Maximum acceleration for extruding value set to 500 in the 'Motion Ability' tab for the extruder.  Seems to be working fine now.  Sorry for any trouble.

closed 2023-01-29 by DarthMaledous

vicpon, 2023-04-13:
@DarthMaledous how did you make the "Motion Ability" tab appear? I don't see it for my printer.

DarthMaledous, 2023-04-14:
> @DarthMaledous how did you make the "Motion Ability" tab appear? I don't see it for my printer.

Sorry for the late response, I just saw this.  Makre sure you have the "Advanced" option turned on. If you look at the screenshot in my initial post, you'll see the slider for it in the top right hand corner.

DarthMaledous, 2023-04-14:
Sorry for the late response, I just saw this. Makre sure you have the "Advanced" option turned on. If you look at the screenshot in my initial post, you'll see the slider for it in the top right hand corner.

⁣Get BlueMail for Android ​

vicpon, 2023-04-14:
> Sorry for the late response, I just saw this. Makre sure you have the "Advanced" option turned on. If you look at the screenshot in my initial post, you'll see the slider for it in the top right hand corner. ⁣Get BlueMail for Android ​
> […](#)

Strange. I do have Advanced on. Maybe it doesn't show up for every printer?

[image]