OrcaSlicer/OrcaSlicer #262: Crashes When Entering Vase Mod

Highlander-bdj, 2023-01-29:
**Describe the bug**
When I select "Spiral Vase" and the click yes the app close.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
Load a Vase Model.  I used this one: https://files.printables.com/media/prints/131488/stls/1261351_850b4f52-1186-4e1f-ab7d-069abe3ab3b6/vaserose_5cm_withbottom.stl
Select Spiral Vase
Click Yes

**Expected behavior**
I expect it to set the necessary settings and enable vase mode.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version BambuStudio-SoftFever_V1.4.4_Mac_arm64

hliebscher, 2023-01-29:
any stl object it is the same M1 arm64 Ventura 13.1

davidja92, 2023-01-29:
i can confirm bug windows 11, softfever 1.4.4 
[image]

hliebscher, 2023-01-29:
its back again

https://github.com/bambulab/BambuStudio/issues/1165

hliebscher, 2023-01-29:
pls check my PR

https://github.com/SoftFever/BambuStudio-SoftFever/pull/264

SoftFever, 2023-01-30:
I just tried it on my machine.
It didn't crash on the Bambu X1C model, but it crashed if I selected Voron.

SoftFever, 2023-01-30:
As a workaround, you can switch timelapse to smooth first before ticking the `Spiral vase` mode.

hliebscher, 2023-01-30:
pls test the fix #265

Highlander-bdj, 2023-01-31:
> @Highlander-bdj <https://github.com/Highlander-bdj> can you Test the fix
> and close the bug?

I am sorry, but I don’t understand the instructions for testing.

hliebscher, 2023-02-01:
you need to compile the source code and addd the fix ...
but you can wait for the next version here for all

Highlander-bdj, 2023-02-01:
I'll wait.  Thank you for looking into it.

hliebscher, 2023-02-02:
> I'll wait. Thank you for looking into it.
> […](#)

Can you close the bug pls?

closed 2023-02-02 by Highlander-bdj