OrcaSlicer/OrcaSlicer #263: Change Flow calibration object shape.

randyre, 2023-01-29:
Noticed you have a flow object that has a circular ring on it but it's not being used, could we use that object as well as make holes off to the sides? 

A smooth flat surface is great for making sure that a smooth flat surface can be printed, but would be nice to make sure that a flat surface with other features also prints well. 

[image]

hliebscher, 2023-01-29:
pls send the STEP file not STL

SoftFever, 2023-01-30:
I haven't found the circle to be useful in my testing.

As a model designed for flow rate calibration, it shouldn't be tested for anything else.
It will remain unchanged hence.

randyre, 2023-01-31:
This is mainly a silk filament issue, but the start and stop of flow, especially on flat surfaces where there is a perimeter or two have a bad ghosting effect that seems to be caused by unequal flow changing the shine of the filament.  I'll attach an example.  

I know it's an extreme use case, but if we either could add it as an alternate flow test or tell me how I can replace the object myself it would be appreciated.
[image]

SoftFever, 2023-02-03:
> 

IMHO this is not a flow issue.
I suggest reducing the speed of silk filaments.

You are welcome to create a post in [Discuss](https://github.com/SoftFever/BambuStudio-SoftFever/discussions)
I'm closing this issue

closed 2023-02-03 by SoftFever