OrcaSlicer/OrcaSlicer #269: Fuzzy skin as Color painting.

PLAmenPRINT, 2023-01-31:
I am printing cone objects (varieties of speaker waveguides) and for more aesthetically finish I am using fuzzy skin.
The point is that there is no need to have fuzzy surface outside the object and it will be great to have a way to chose which surface to be with this effect like the colors with the AMS.
[image]

Is it difficult to be implemented?
"Fuzzy skin Only Inside" will be simple alternative?
Thank You for the attention
Best Regards

DmncS, 2023-08-16:
I talally agree to your issue, it would be great to have it paintable, like colors.

closed 2023-11-25 by github-actions[bot]

netweaver1970, 2024-02-14:
another upvote here. For esthetic reasons I would like the fuzzy skin on specific (mostly outside) surfaces but the build-in function treats too many surfaces as outside. 
So either that function needs to be made a lot smarter (very hard probably as it might be object specific how inside/outside would be humanly determined) or a manual/assised fuzzy skin paint option (prefered) is needed.
Thanks for looking into it.

jilleb, 2025-02-06:
https://github.com/TengerTechnologies/Fuzzyficator adds this feature, by using a 2nd filament, which can be painted on and then should print as fuzzy. So far I haven't got it to work as paint-on feature.