OrcaSlicer/OrcaSlicer #270: Icon size is wrong on MacOS

Guilouz, 2023-02-01:
Compared to original Bambu Studio, icon is oversized.

On dock when is not launched, icon is oversized :

[image]

On dock when is launched, icon is correct :

[image]

In list mode, icon is oversized too :

[image]

SoftFever, 2023-02-01:
Thanks for the feedback.
Will fix it in the next update.

Guilouz, 2023-02-21:
@SoftFever Just see your last update and it's not fixed.

closed 2023-11-25 by github-actions[bot]