OrcaSlicer/OrcaSlicer #273: On Ubuntu when sending a single or all plates the program crashes with a segmentation error

RobRobM, 2023-02-01:
**Describe the bug**
If I use "Send" or "Send all" instead of "Print plate" the program crashes and I get a segmentation error.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Prepare'
2. Click on 'Send' or 'Send all'
4. See error/program crashes

**Expected behavior**
Being able to send plates to X1C to be able to print later.

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Version 22.10

broncosis, 2023-02-02:
I can confirm this behavior on Linux Mint 21.

Geoff-S, 2023-02-04:
Also have this on Ubuntu 22.04

broncosis, 2023-02-06:
Just on a fresh install of Mint 21.3   was on 20.4 not 21  and the send function works 
My guess is possibly a newer network driver from bambu lab

SebiPanther, 2023-02-06:
I like that theory - lets check! :D
[image]

> Just on a fresh install of Mint 21.3 was on 20.4 not 21 and the send function works My guess is possibly a newer network driver from bambu lab
> 
> if you remove you config files for bambu studio it should download the network driver if I had not just flushed my old os I would have tested this

I like that theory - lets check! :D
[image]
`Build-ID: ab3f6a3d1a456eb18c1362c42c4321f53bb3f014`
`ebf126e00d3c9da0de77b9e4ca88297223776a92  libbambu_networking.so`
After renaming that file libbambu_networking.so BambuLab (1.4.4 SoftFever fork) shows the download dialog:
[image]
And the new file is different:
[image]
`Build-ID: 286b3578b3a213e9f2538002a6b13f3f2bd75ecc`
`a199ac6d7ecd9f869c5963376f17855daa93d40a  libbambu_networking.so`
Does the send file work? **Yes it does!**
[image]

**So delete ~/.config/BambuStudio-SoftFever/plugins/libbambu_networking.so and let Bambulab automatically download the new version.**

Rant about Bambulab: I really have to be careful with my words about BambuLab. Not only its bad practices to download and execute code from the web without consent of the user. They also have a not working update mechanism in that part of there crapware.
That machine would not be unusable without Forks like this - thank you for your support guys!
For that kind of money that support should come from BambuLab! I mean: They even found a "clever" (nearly illegal) way around to not refund people in Europe. I'm now stuck with that crap and i only can hope that people you guys are clever enough to fix the issues. Even that shitty company will benefit from it.

broncosis, 2023-02-07:
@RobRobM   see above post this is good news Sebi confirmed it

RobRobM, 2023-02-07:
@broncosis I was trying to delete ~/.config/BambuStudio-SoftFever/plugins/libbambu_networking.so but it was not in my plugins directory. I started Softfever and it started downloading the new network driver, weird. And "Send" and "Send All" is working.

So I did nothing but it has started working.

Thanks for replying and all the work!!!

closed 2023-02-07 by RobRobM

SebiPanther, 2023-02-08:
> Still having this problem after updating libbambu_networking.so (mdsum 920305a5199db4f7460eb4fa12f26782 ) Happens with Print, Print All

Sorry to hear that - but i think its a different problem, since this thread is about sending and not printing. Printing worked with both version in my case. Maybe try to delete (or move away) the `~/.config/BambuStudio-SoftFever/` folder (all settings get lost! be aware!). It helped me in other cases. Your md5sum seems to be that same (new) version i got yesterday (shasum a199ac6d7ecd9f869c5963376f17855daa93d40a ).
Good luck!

Geoff-S, 2023-02-08:
OK Thanks,
Didn't realise that this referred to sending only. 
I'll do as you suggest and see what happens.