OrcaSlicer/OrcaSlicer #280: Generating gcode stage of slicing stalls with 100% CPU utilization

Elbau, 2023-02-03:
During slicing of any file, including the built-in calibration models, the application hangs at 80% (generating gcode) and the CPU utilization goes to ~100% (~950% on my 10 core CPU). Slicing will succeed after an extended duration (12 seconds for the flow calibration model, 55 seconds for a simple part).

This happens with any file I try to slice, more complex geometry hangs for longer than simple geometry.

This does not occur for me with BambuStudio or PrusaSlicer with the same geometry. 

I'm using version 01.04.02.13
My machine is a Mac, M1 Max w/ 32gb RAM, MacOS 13.1 (22C65)

hliebscher, 2023-02-05:
at my M1 8 Core os 13.1 it works very fast like Prusa or any other slicer

Elbau, 2023-02-14:
After testing the precompiled executable, this seems to be an issue with my local build. I must have something misconfigured in my build environment.

closed 2023-02-14 by Elbau