OrcaSlicer/OrcaSlicer #288: Spiral Lift always uses G3 command

javales74, 2023-02-06:
SpiralLift always uses G3 command to make the hop, even if Arc Fitting is disabled, I think if "Arc fitting" is disabled should uses G1 command.

[SpiralLift_bug.3mf.txt](https://github.com/SoftFever/BambuStudio-SoftFever/files/10669923/SpiralLift_bug.3mf.txt)

hliebscher, 2023-02-06:
Arc fitting is in part settings and SpiralLift in extruder settings

i found this comment from bambulabs 
//BBS: todo: check the arc move all in bed area, if not, then use lazy lift

is G3 for your printer a problem? or why do you need G1 Commands?

javales74, 2023-02-07:
Not a big problem, I use Klipper and need to explicit activate this function and uses a lot of resource of my RPI3, report this because I discover this bug due arcs are disabled in my Klipper configuration.
Only tried SpiralLift for testing purposes, NormalLift works OK for me.

closed 2023-02-10 by javales74