OrcaSlicer/OrcaSlicer #294: Spiral Vase mode crashes Bambu Studio

Fl0ken, 2023-02-08:
**Describe the bug**
Loading in any file (STLs) that should be in vase mode and then clicking Spiral vase crashes the application

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Load file
2. Click on Spiral Vase
3. See error

**Expected behavior**
Bambu Studio slices the file

**Screenshots**
Error Code: 
20:16:54: BambuStudio got an unhandled exception: boost::bad_any_cast: failed conversion using boost::any_cast
20:16:54: BambuStudio got an unhandled exception: boost::bad_any_cast: failed conversion using boost::any_cast

**Desktop (please complete the following information):**
 - Windows 10
 - 21H2

hliebscher, 2023-02-08:
As a workaround, you can switch timelapse to smooth first before ticking the Spiral vase mode.

we have fixed thisin next version

Fl0ken, 2023-02-08:
Version is 1.4.4
Smooth timelapse works, but the prime tower needs to be enabled, otherwise it's crashing.

hliebscher, 2023-02-09:
you have to compile yourself or wait for the next 1.4.5 
we have fixt this

[image]

closed 2023-02-10 by Fl0ken