OrcaSlicer/OrcaSlicer #295: Mainsail displays wrong slicer name

Ibot-11, 2023-02-09:
When using Mainsail the table column for the slicer shows Superslicer.
Also a new name for this fork would be really nice.To say “Bambustudio SoftFever Fork” every time anyone asks you what slicer you are using is a bit annoying.
This fork already has so much more features than Bambstudio, that it really deserves a own name.
[image]

hliebscher, 2023-02-09:
this is not a bug!  
hack-fix: write fake slicer info here so that Moonraker will extract thumbs.

SoftFever, 2023-02-11:
It's a temporary solution.
Moonraker can now recognize this slicer properly
I had a PR that had already merged into Moonraker mainstream a few months back.

The only thing left is to submit a PR to [`object canceling`](https://github.com/kageurufu/preprocess_cancellation) post-processing so that it recognizes this slicer.
I properly will implement preprocess in the slicer directly though.

closed 2023-11-25 by github-actions[bot]