OrcaSlicer/OrcaSlicer #296: Overhang speed is not respected on painted/coloured parts

rovili, 2023-02-09:
**Describe the bug**
Overhang speed is not respected on painted/coloured parts

**To Reproduce**
Take a part with overhang and color the overhang

**Expected behavior**
Overhang speed should also be applied on coloured parts

**Screenshots**
With different color:
[image]
[image]

With the same color:
[image]
[image]

**Desktop (please complete the following information):**
 - OS: macOS
 - Version 1.4.4

hliebscher, 2023-02-10:
Can you Send the Link to stl file?

rovili, 2023-02-10:
Hi hliebscher, 3mf and stl file-type are not allowed to be attached in his repo. 

Here are the links to STL and 3mf:
https://github.com/rovili/BambuLab/blob/main/leaf-coasters-pot-sculpted.3mf
https://github.com/rovili/BambuLab/blob/main/leaf-coasters-pot-sculpted.stl

rovili, 2023-02-15:
@hliebscher any updates on this?
I noticed that it is also glitchy on not multicolor prints, some overhang (50°+) just don't get any "slowdowns" - but they do in Bambu Studio.

hliebscher, 2023-02-15:
@rovili you can add 3mf as zip ;-)

hliebscher, 2023-02-15:
it works for both

rovili, 2023-02-15:
...I can reproduce it with more than one object. While it is totally fine with Bambu Studio. 
So something must be glitchy. Also color independent.

closed 2023-02-20 by rovili