OrcaSlicer/OrcaSlicer #298: Segfault on launch on Fedora 37

shellixyz, 2023-02-10:
**Describe the bug**
The program segfaults on launch just after showing the main window

`fish: Job 1, 'bin/bambu-studio' terminated by signal SIGSEGV (Address boundary error)`

When run with Valgrind it doesn't crash.  [valgrind_output.txt](https://github.com/SoftFever/BambuStudio-SoftFever/files/10706069/valgrind_output.txt)

**To Reproduce**
Steps to reproduce the behavior:
1. Launch revision 1.4.2 AppImage on Fedora 37

**Desktop (please complete the following information):**
 - OS: Fedora 37
 - Version 1.4.2

shellixyz, 2023-02-10:
Actually it does crash after a little while after launching without doing anything. I guess it takes a little bit more time just because it is slower when launched with Valgrind. Look like a NULL pointer dereferencing.

[valgrind output.txt](https://github.com/SoftFever/BambuStudio-SoftFever/files/10706164/valgrind.output.txt)

andypiper, 2023-02-14:
I'm also seeing a crash on Fedora 37, running the AppImage. How are you invoking it - local build from source and then valgrind on that? Just curious if we are seeing the same issue.

shellixyz, 2023-02-18:
I'm running the AppImage with `--appimage-mount` then running `valgrind bin/bambu-studio` in the app mount dir

rehashedsalt, 2023-04-19:
Able to replicate this on Fedora Kinoite 38 over here.

brychanrobot, 2023-07-27:
+1. Any updates on this?

ikilledmypc, 2023-07-28:
I managed to compile Orca slicer using the instructions from Bambu studio for fedora and can generate a working appimage for fedora by invoking the src/BuildLinuxImage.sh from the build folder after building. 2 issues though the cam doesn't work and there's a dismiss able javascript error on startup but other than that everything seems to work

shellixyz, 2023-08-01:
I tried building it myself it does work on F37

jumski, 2023-09-18:
same on Manjaro (kernel 5.15.128-1-MANJARO), orca slicer 1.6.3

closed 2023-12-26 by github-actions[bot]