OrcaSlicer/OrcaSlicer #300: "Print All" Button

tabiznet, 2023-02-10:
**Is your feature request related to a problem? Please describe.**
Sending prints to 15+ Bambu printers is a major pain. I run a farm so I usually print the same thing across all of my printers at once.
**Describe the solution you'd like**
An option in the Print button drop-down that sends the sliced plate to every printer connected to my Bambu account and triggers prints.
**Describe alternatives you've considered**
I tried setting something similar up with MQTT, NodeRed, Home Assistant, and an FTP server but it's too buggy to be reliable.
**Additional context**
Add any other context or screenshots about the feature request here.
A nice optional feature but not necessary for my use case, would be after selecting "Print All" a menu appeared where you could select which printers you want to send it to. Like a checkbox or something with a "Select all" button.

hliebscher, 2023-02-11:
The easiest way is to use the post-processing script to send the G-code file sequentially to the printers.

dojo90, 2023-03-29:
@hliebscher Did you make any special settings so that you could ftp the gcode file to the printer?

With me the ftp connection always breaks down, see https://github.com/bambulab/BambuStudio/issues/1404

*I have a P1P*

closed 2023-11-25 by github-actions[bot]